grbl/grbl #1455: Possible ID 10T Error

Ajb2k3, 2018-10-28:
I'm trying to control a turntable driven by one stepper but I'm struggling to get in dialed in.
The stepper is connected to the "Z" port however 
`G91 X0Y0Z1 F30`
isn't moving it one step or 1.8 degrees clockwise.
Can you tell me what I'm missing?

mayhem2408, 2018-10-28:
Z1 is not telling GRBL to more 1 step, it is telling it to move one millimeter. You need to make sure you $102=1 to tell GRBL that there is 1 step per millimeter.

Ajb2k3, 2018-10-29:
Can you point me to a referance on it please?

mayhem2408, 2018-10-29:
>           Can you point me to a referance on it please?

To understand Gcode itself you can have a read of
[](https://en.wikipedia.org/wiki/G-code)

and for the GRBL settings you can read
[](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)

Ajb2k3, 2018-10-30:
> > ```
> >       Can you point me to a referance on it please?
> > ```
> 
> To understand Gcode itself you can have a read of
> https://en.wikipedia.org/wiki/G-code
> and for the GRBL settings you can read
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Thank you. 
I thought it was an ID 10 T error all along (idiot).

closed 2018-10-30 by Ajb2k3