grbl/grbl #1458: M5 Question

barnold96, 2018-11-02:
I'm adding a laser engraver to my Shapeoko2 CNC.  

I downloaded the 1.1f hex file and uploaded it to my Arduino UNO using XLoader.  I had tried to download the grbl-1.1f.20170801.zip and compile with Arduino IDE but kept getting errors.  Going the hex route loaded without any problem.

The issue I'm having is M5 doesn't do anything.  M3 turns my router(spindle) on and M4 turns it off.  It's the same condition whether I have $32=0 or $32=1.  

The post processor I'm using with VCarve for a test file calls M4 at the start and the laser seems to work correctly.  The issue is with CNC gcodes where M5 is called at the end of a program to stop the spindle.

Is there a setting I'm missing?

chamnit, 2018-11-12:
M5 turns off the spindle. M4 enables Grbl's dynamic laser mode. This mode is unique and doesn't fire if it's not moving. See the Grbl wiki pages on how the laser mode works. (Not on this site, but the Gnea/Grbl site).

closed 2018-11-12 by chamnit

barnold96, 2018-11-27:
M5 does NOT turn off the spindle under grbl 1.1f.  I have run it several times and it doesn't work.

barnold96, 2018-12-04:
The pins are in the correct configuration.  Since updating to 1.1f, repeated attempts to test spindle on and off give me the same result.  I'm using the PWM on pin 11 to drive a laser.

I was on grbl 0.9j prior to adding the laser and updating to 1.1f.

bjrenken, 2020-02-26:
> The pins are in the correct configuration. Since updating to 1.1f, repeated attempts to test spindle on and off give me the same result. I'm using the PWM on pin 11 to drive a laser.
> 
> I was on grbl 0.9j prior to adding the laser and updating to 1.1f.

I'm having the same issue. Did you find a solution?