grbl/grbl #1462: Query: Homing and Work Space

AcrimoniousMirth, 2018-11-16:
Hi all,
Sorry if this has been asked before or is a poor place to put it.
I’m wondering... if I were to manually jog the head to an unknown space in the machine coordinates, Zero that as the work coordinates, then home the machine, would it then use the distances travelled to home as the work space offset from the machine space?

109JB, 2018-11-16:
What youdo is you home the machine on start up and this gives it a repeatable reference point.  Then,  go to your desired location I'll on and zero the work coordinates.  Then you will have the offsets set. Then if you shut down and re home,  your work coordinates will already be set to the accuracyof your limit switches

AcrimoniousMirth, 2018-11-16:
@109JB Thanks, I get (and have implemented) that. I just think there's something to be said for quickly moving the router to its work 0 by hand, moving the bit down to touch the material by hand (and thus with more control and accuracy than jogging it by software), then homing from there to get its exact coordinates without using a "best guess" incremental method like jogging it to smaller and smaller units and saying "yeah, that's probably close enough. I think it's touching but I'm afraid jogging it slight;y further will push the bit into the surface".

AcrimoniousMirth, 2018-12-13:
@MeJasonT hi, thanks for your reply.
Actually I do understand how and why homing is done and I’ve been successfully running my machine for a while now. My second ever test was a successful 3D cut requiring tool changing, which I was very exited about.
I think you’re actually just missing the point I was trying to make and the logical reasoning I was coming from. It just seems to me that for a first setup, this method would be quite helpful.

109JB, 2018-12-14:
As I understand it Grbl homing doesn't work that way as it doesn't really track position while homing is occurring and therefore can't get a measure of how far it moved during a homing cycle. 

> "yeah, that's probably close enough. I think it's touching but I'm afraid jogging it slight;y further will push the bit into the surface".

As for finding an edge accurately and quickly. there are probes, tool setters, edge finders (electronic and mechanical), etc, that can find an edge or set a z-position more accurately than the feel you have by hand, and they don't do any damage to the work piece.

AcrimoniousMirth, 2018-12-14:
@109JB doesn’t it? Isn’t that how the soft limits work? If it’s aware it’s moved further than the physical length it should move, it shuts down?
Those tools are great but not everyone is going to get them. They seem a bit fddly and complex to set up, at least the XY probes. Z is simple enough.
It’s okay if there isn’t an implementation, I’ll live. I just think it makes a whole lot of sense for initial setup.

MeJasonT, 2018-12-14:
Automating the process of finding the work zero position is something i've often thought about, I guess probing to locate the work piece and then responding to first contact for locating the edge would be possible.  Unfortunately its more a function of the software application than it is GRBL.  It would be nice to click on one button to store XYZ locations when a part was manually set up,  I'm still in the realms of OCD when it comes to entering in the values and getting them right, humans don't naturally think in inverse coordinates.  Unfortunate it this is also a function of a would be software application.
AcrimoniousMirth - there are a whole load of of fantastic functions that could be written within the limitations of the GRBL Gcode command set, perhaps  a function to be written which was added to the post processor, or simply a program that was run before the job program.  the work coordinates would still be saved and would not change until a power reset was performed or they were intentionally overwritten.

If someone is a good coder and wants a challenge then Gerrit Visser is looking for someone to take on Grbl Panel, I've thought about it as i would love to add more functions to it but unfortunately Visual Studio frustrates the hell out of me.  I always have hassle with references and code not compiling, I especially hate Python and java, i'm more of a double click to load sort of a person.  Entering Code to get a program to start I grew out of in 1986.

AcrimoniousMirth, 2018-12-14:
Whereas I’m coming at it from the point of view of knowing pretty much what I want, how to work it and why I want it but almost none of the coding skills necessary to implement it!

109JB, 2018-12-14:
> @109JB doesn’t it? Isn’t that how the soft limits work? If it’s aware it’s moved further than the physical length it should move, it shuts down?

Soft limits know where the machine is at and act accordingly to not exceed those limits, but the homing cycle is different and is not "normal" operation. During "normal" operation Grbl continuously tracks the location. Homing is a special cycle to FIND the home position so that it knows where it is from that point on.  In very basic terms the homing cycle works by turning the motors at a set rate until the limit switch is triggered. During this homing move it doesn't monitor position, it just keeps sending step pulses until the switch is triggered. This is because it is going to reset the machine zero position anyway at the end of the homing cycle so there isn't a need to monitor position during this homing cycle. It could have been written to monitor the position, but it just wasn't written this way. I asked a while ago about this and was told that because of the way the homing cycle was written it is not trivial to change it.

In an open loop system, the only way to manually move the machine is to de-energize the steppers and them move the machine manually. At that point Grbl no longer knows where it is at. The only way for Grbl to maintain awareness of the current position is for Grbl to command EVERY move that occurs, and the only way for homing switches to do any good is to home first before doing anything else, and do everything from the point of homing on by using Grbl to move the machine.

If you have a low power (axis drives) machine that would not be damaged by running into a physical limit at top speed, or a machine with limit switches far enough from the ends of travel for the machine to stop before damage, then you can already do what you want to do by simply making sure the work coordinate offsets are zeroed ($RST=#), disable soft limits, then move your machine manually to where you want zero to be, then power cycle the arduino. On the power cycle, MPOS is set to 0,0,0 and with work coordinate offsets set to zero your G54 will now be 0,0,0 as well.

If you do what is described in the previous paragraph you must be aware that homing is now useless and if for some reason you lose position while making the part you will have to re-zero somehow. Imagine if you set z=0 this way to the top of the part and then during the making of the part that feature is machined away. You now have to find another way or location to use to set position.