grbl/grbl #1463: Hard Limit Triggered when leaving Home

AcrimoniousMirth, 2018-11-16:
Hi,
So I set up the work space, press "Start" in CNC.js and the first thing it does is homes (this is code generated by F360). Once homed it tells me to do an M6 tool change then press "resume". So I press resume and it tries to move away from home but triggers a HARD LIMIT immediately, so I can't move away from home other than manually moving it some distance away and then pressing resume, which obviously cancels out any accuracy and repeatability I may have had.

109JB, 2018-11-16:
The homing command should not bein the program.  Are you sure itis a $H that isin the program file?

My guess is it isn'ta $h but something like a G53 X0 Y0 Z0. If this is the case just change the X Y and Z coordinates to something slightly negative to keep it off of the switches.

jahnj0584, 2018-11-16:
This is just a G28 or G53 command in your fusion post that you SHOULD
DISABLE. There is no reason a machine always has to go to machine origin at
the start of a file. This setting causes so many issues for so ma y
newcomers.

AcrimoniousMirth, 2018-11-16:
@jahnj0584 @109JB  Hi, yes, that's what I ended up doing. It was ``G28 G91 Z0`` which ended up just homing them all. I changed it to ``G28 Z`` and it works fine now. I'll probably remove it entirely. Just a pain to have to delete it for every new operation.

On a related note, I was doing my first test that involved a tool change. I faced the part, it then raised out of the stock and paused so I could change the tool, so I did. It told me to run an "M6 Macro" but GRBL told me that's an "incorrect number" so I did it manually, zeroed the new bit and pressed resume. It then proceeded to start the cut from the beginning again. 
I said "whatever, something to sort next time" and left it to run through that and then through what it was meant to do, but it then just randomly stopped halfway through that process. I checked the CAM tool path preview and it showed the bit stationary and just spinning in the middle of the tool path, and the play and pause buttons were greyed out, if memory serves. Any idea what caused that? And how should I do a tool change properly? I understand the principle but I'm thinking the code may be more involved than I assumed!

Thanks for the help :)

jahnj0584, 2018-11-16:
Seperate file for each tool.

G28 is an option in your fusion post. You don't have to do it every time.