grbl/grbl #1465: Motors does not work

RafatCAD, 2018-11-19:
I don't know where is the problem , motors work when I use repetier host, but does not work in Grbl all thing I think it is correct , but I don't know where is problem ,

RafatCAD, 2018-11-19:
[image]
[image]
[image]
[image]

svenhb, 2018-11-19:
You didn't enabled Homing - as the response says: "Homing cycle failure. Homing is not enabled"
Read here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration
$22 will enable homing.
But for first use you should avoid homing, first setup directions, end switches and so on.

svenhb, 2018-11-19:
You use version 1.1g. The latest official release is 1.1f... What is the difference?

RafatCAD, 2018-11-19:
1.1g this is for mega

chamnit, 2018-11-19:
1.1g is a bug fixed version on both the 328p and Mega. Mostly minor bugs.

svenhb, 2018-11-19:
You should check the wiring of the motor-shield. I remember a similar looking shield were clk and dir was switched (compared to usual Arduino grbl pinout).

svenhb, 2018-11-19:
The one I mean is from banggood (CNC-Shield-V4-With-Nano-3_0-A4988), if you follow the conductor track you find "dir" (pin 1) of one driver connected to D2 ="stp" of default grbl.

svenhb, 2018-11-19:
For the mega, the default X_dir is on Pin 30: https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega
But for the Ramps 1.4 shield, X_Dir is connected to A1: https://reprap.org/mediawiki/images/f/f6/RAMPS1.4schematic.png
Probably that is the problem.

RafatCAD, 2018-11-21:
I test my project on UNo everything works, I try to put cnc sheild on mega it also doesn't work . 
I think mega have something must change  on CPU .h .

RafatCAD, 2018-11-21:
I use mega because I need at least 4 motors

terjeio, 2018-11-21:
Perhaps the motor enable signals are incorrect? The RAMPS board has a separate enable signal for each motor (active low).

RafatCAD, 2018-11-22:
Problem solved, but I have  inquirie , the probing in UNo on A5  pin but in mega in A15 (thermal sensor 2) . This thing is Wright or not

closed 2018-11-22 by RafatCAD

svenhb, 2018-11-22:
I figured out same relation in the links posted above.

mkaluza, 2018-12-11:
@RafatCAD Hi,
I've just got my hands on RAMPS 1.4 only to find out it's not supported by grbl. Can you post the changes you made somewhere?

mkaluza, 2018-12-11:
ok, nevermind. It appears you just need to recompile grbl-mega. I was looking in the wrong repo.