grbl/grbl #1467: ALARM: Hard limit [Reset to continue]

ramimiro, 2018-11-20:
Hi
My problem is to limit switch when trying to succeed in the task   

Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
>>> $H
ok  

When I want to print, stop when you touch any end of the switch 

Skipping command #1
Skipping command #3
>>> G20G90G40
>>> G0Z0.2
>>> T0M6
>>> G17
>>> M3
>>> G0X4.264Y2.5334
ok
ok
>>> G1Z-0.025F20
>>> G1X4.2547Y2.4316F40
>>> G3X4.2516Y2.3896I1.0665J-0.0976
error: Unsupported command
ok
>>> G3X4.248Y2.304I4.4118J-0.2289
ok
ok
ok
>>> G3X4.2459Y2.2183I5.6542J-0.1836
ok
ok
>>> G3X4.2452Y2.1401I4.7085J-0.0815
ok
>>> G2X4.2433Y2.0368I-2.9542J0.0018
ok
>>> G2X4.2399Y1.9921I-0.5557J0.0198
ok
>>> G2X4.2284Y1.9389I-0.277J0.0323
ok
>>> G2X4.2145Y1.9129I-0.0782J0.0252
ok
>>> G2X4.1921Y1.8951I-0.0498J0.0397
ok
>>> G2X4.154Y1.8841I-0.0526J0.1103
ALARM: Hard limit
[Reset to continue]

chamnit, 2018-11-20:
Please see the wiki FAQ about electromagnetic interference.

closed 2018-11-20 by chamnit

ramimiro, 2018-11-20:
Thank you my friend but I looked they did not have a similar problem