grbl/grbl #1470: Drauing square i stead of circle

CFRLinux, 2018-11-27:
Hi ..
has a small problem,
is building my first CNC machine,
and the teases little
uses an arduino mega, and universal g-code sends
has full control over the three axes,
but when I import a file with the text below,
and virtue it's a nice circle,
but the machine draws a square that stands on one spit,

Whats wrong

G17 G20 G90 G94 G54
G0 Z0.25
X-0.5 Y0.
Z0.1
G01 Z0. F5.
G02 X0. Y0.5 I0.5 J0. F5
X0.5 Y0. I0. J-0.5
X0. Y-0.5 I-0.5 J0.
X-0.5 Y0. I0. J0.5
G01 Z0.1 F5.
G00 X0. Y0. Z0.25

jahnj0584, 2018-11-27:
Use ncviewer.com to confirm your code

CFRLinux, 2018-11-27:
tried,
a nice circle, but still a square on the CNC

Norwegian-Gadgetman, 2018-11-27:
Your code is rather messy.   
What did you use to generate it?

There's a lot of lines that lacks the G0/G1/G2 command. 
From what I can see by glancing at the code, it's drawing the circle in four 90degree arcs. Some commercial machines have a limitation when cutting circles, but GRBL is quite capable of doing 360degree cuts. 
So a single G2 command where the ending coordinates is the same as the starting coordinates will work just fine. (And I and J of course references the center)

CFRLinux, 2018-11-27:
it came from. https://github.com/Protoneer/grbl-master/wiki/G-Code-Examples

CFRLinux, 2018-11-27:
Have tried different pages on the internet,
which claims they can make G code for circles
but nothing works :-(
Have someone, a little example that you know works

109JB, 2018-11-27:
That code should work just fine and it should not make any difference in the final shape. The only thing it might do is result in a slowdown at the junctions, but I think the planner is good enough that this won't be noticeable.

If you are getting squared off features using that g-code then my guess would be that your machine has a lot of backlash or flex and what is happening is when each axis reverses while going around the circle the backlash is making the shape odd. It usually looks something like this:

[image]

If that is what you are getting then it is backlash. Grbl doesn't currently have backlash compensation, so the only option is to make the machine so that it has as little backlash as possible,

CFRLinux, 2018-11-27:
it has nothing to do with backlash to do,
and it's a very nice square

CFRLinux, 2018-11-27:
The X and Y axes run at the same speed
they should not do that on a circle

chamnit, 2018-11-27:
What are your $$ Grbl settings and $I build info output?

chamnit, 2018-11-27:
Your arc tolerance setting is extremely high at 400mm. It should be less than 0.01mm to trace an accurate arc.

closed 2018-11-27 by chamnit

CFRLinux, 2018-11-27:
there was the problem :-)
thanks :-)