grbl/grbl #1471: Finished build, CNC not working :P HELP

Seajay67, 2018-11-28:
Hi.
I've just finished the mechanical and electrical build for a large OX-CNC from Makerstore.
We are running arduino uno, GRBL0.0j, CNC shield v3.0 with TB6600 stepper drivers and adapter cables to run from the CNC shield to the drivers.
I have modified the config to take into account that the spindle needs to be disabled to work with CNC shield v3.0 and I have specified the OX-CNC settings.
GRBL uploads to the Arduino okay. GRBL controller see's the arduino, and the config is displayed (see below) We have limit switches at each end of all 3 axis. NO. I have used a multimeter to check all the wiring to switches and connectors to the stepper drivers.
And all that happens when I turn it all on is the steppers lock up, and i  get errors back from the GRBL. Can't get it to jog, can't get it to home...... 
First time I'm building a CNC, for my school.  
help!
[image]

[image]

Seajay67, 2018-11-28:
Errors I get include:- ALARM:homing fail, error undefined feed rate (when trying to jog using controller)
no movement anywhere 📦

zentoolworks, 2018-11-28:
Try to run with a feedrate
G1 X10 F100

Seajay67, 2018-11-28:
Tried, got nothing

zentoolworks, 2018-11-28:
what do you see after connection established? or just type in "?" and post response here.

manisaranbala, 2018-11-28:
I think pul and dir  change

Seajay67, 2018-11-28:
On connection establish i get:- 
[image]

manisaranbala, 2018-11-28:
Dir puls wiring check
My WhatsApp number India +919786876586
I don't speak English. only chat  WhatsApp

Seajay67, 2018-11-28:
Checked and fixed DIR puls wiring (had them backwards)
Still not working after rewire.

zentoolworks, 2018-11-28:
Is the grbl in idle state while you sending command. If it is in Alarm state, you will need to unlock it first.

Seajay67, 2018-11-28:
With assistance of manisaranbala (thank you!) I have rewired the control circuitry to the TB6600. I now have movement. I need to check that wiring again as the movement is only in the one direction. on all 3 axis (x,y and Z) so with for example G01 X-10 F100 it moves in one direction, but G01 x10 F100 still moves in the same direction.

MeJasonT, 2018-12-13:
It sound like your stepper motors may be wired incorrectly.

drf5n, 2025-02-26:
> It sound like your stepper motors may be wired incorrectly. where your phases of the motor should be A+A-B+B- you may find you have A+A-B-B+ for example effectively you get half the steps. with the second pair reversed when signal is ran in reverse that pair runs half a step in same direction.

Reversing a single coil will only reverse the CW/CCW direction.  If you mistakenly cross the coils, like A+B+A-B- or A+B-A-B+ you'll get unidirectional movement because the coils will be powered strangely with odd Brake signals.