grbl/grbl #1477: How to enable door safety?

mboldrini, 2018-12-11:
How activate the door pin?

the button should be set to NO or NC?

hoesslin, 2018-12-11:
dude, whats about „tryout yourself“??

sorry guys, that i respond to that! but iam sick about some kind of people and their questions. 

a lot of questions would not be questions anymore if some people wouldnt be so lazy and try out theirselves!

if you want to play in the DIY league, then act like a tinker! 

the founder of this grbl provides this sources for free - this dude is not your nanny solving your problems because you are to lazy to read (RTFM) or you have 2 left hands!

if you are not able to play with DIY ware - buy a dog or play lego!

MeJasonT, 2018-12-11:
Hoesslin, your not helpful either.  before you rant you could have suggested where he might look.  Not everyone is an expert like you.

MeJasonT, 2018-12-11:
mboldrini - try 
https://github.com/grbl/grbl/issues/713
https://github.com/gnea/grbl/issues/241
https://github.com/grbl/grbl/issues/1267
https://github.com/grbl/grbl/issues/1056

hoesslin, 2018-12-11:
i am far away from being an expert! but i dont like this selfish behaviour - not wasting OWN time with RTFM or trying out, but wasting others time! 

you found and posted answers to his question. you are dedicated and spent time for him - you did his job!

but "he" - ok, i catched HIM now - would have found your posts too if he had searched!

anyway - whats the best answer for the question he asked? try it out - it takes 5 minutes of your time - then you know whats the best way!

i also spent so many hours in getting this "shit" K40 laser working - this endstops made me crazy and grey hairs - but i found answers for any question by reading this threads or searching google! 

THATS the difference!

MeJasonT, 2018-12-11:
I really hate narcissistic uneducated arrogant individuals who feel its their responsibility to make others learn the hard way, as if they enjoyed similar treatment when they were starting out.  If you look at the big picture Sonny and others have provided their time and provided resources free to the WORLD for little reward.  As a result many people can now program (even if its in a limited fashion).  They can build their own CNC machine and 3D printer.  Why do you resent spending 30 seconds pointing someone in the right direction.  If you really don't like helping people then stay away from forums, I can guarantee people will ask questions that may seem simplistic to you.  I have met people in Mensa who are not even capable of washing or looking after themselves and yet they have the highest IQs in the world.  Be more humble dood, you will find life more fulfilling.

Rob2029, 2018-12-11:
Everyone  in some point of their learning process didn't know what is best
for their machine and if he doesn't want to try out before asking that's a
lazy man right there BUT, no one should be ranting about some dude's
question, just don't reply at all if you're not going to help. I get your
point but we need more nice people here. That's my opinion

hoesslin, 2018-12-11:
"narcissistic uneducated arrogant individuals" - well, i dont feel addressed ;)

no, please try to understand my bad english - maybe i can express myself in correct english.

i tried to explain, that every humans on this world are same - no matter what race, sex, religion or what else!

and it is our all fucking responsibility to respect and appreciate each others life, time and personality!

we all have less time, we all - me too - like to help people, thats the spirit of DIY and this communities, but this expectant attitude to give others the responsibility for my problems because i have better to do then searching threats and google is not correct!

i read a lot of questions here and elsewhere and i also often think - fuck, what a tricky question but you also can read out the difference. these tricky questions sound like "i tried xyz, then i searched xyz - but it did not solve my problem - please help, what should i do"

this is what i tried to explain. asking questions you could solve by your own just by trying out - it takes 5 minutes - but you dont is disrespectful. it means "my time is more worth then your time, so YOU solve MY problem". 

to open a door - should i press the doorhandle up or down...try it out!

if he had written - "guys the door pin, i tried this and this and searched here and here, but i dont work" -  i never opened my mouth!!

i dont have anything against the thread opener - he is just an example for a typical behaviour.

topic closed for me - this is my understanding of respect.

Rob2029, 2018-12-11:
No one is obliged to solve anyone's problems, as I said, don't answer if
you're not going to help, this is a learning and friendly place, grow up...

chamnit, 2018-12-11:
@mboldrini : You have to enable the safety door feature in config.h and recompile. There is an invert pin option as well to set NO or NC. 

Locking this thread because it got nasty quickly.

closed 2018-12-11 by chamnit