grbl/grbl #1478: Homing Fail, switches work

Napalm1432, 2018-12-11:
So I just finished my CNC build, and am configuring GRBL, but it keeps giving me this error:
"An error was detected while sending $H:ALARM: Homing fail."
I checked if all my endstops actually work with $23 and they all give the right output.
Also when I set hard limits to 1, the axes will stop moving when hitting the corresponding endswitch.
But anytime I try the $H command, it will move up the Z till it hits the switch, then spits out the error.
Been looking around and trying all kinds of configurations, but all fail so far...
Hope someone can help :)

hoesslin, 2018-12-11:
@MeJasonT  ... "**Been looking around** and **trying all kinds of configurations**, but all fail so far..." 

understand now, what i tried to say?

MeJasonT, 2018-12-11:
ok If i remember right for homing to work you have to have hardware limits enabled.  software limits are ignored whilst homing.  $27 you may not want to pull you machine of home position but i would give yourself 3 to 5 mm just for the sake of testing - is your machine homing in the right direction.  Industry CNC standard practice is far top right corner for X&Y with Z Up.
These were a couple of things that got me when I first ran my machine.
I was going to ask what version of GRBL you were running but i see its an older version as there are no settings for laser control.
Are you using the correct wiring for the version of GRBL firmware before version 0.9 the limit pin for Z and Spindle direction were different to that of version onward

[grbl wiring.docx](https://github.com/grbl/grbl/files/2669123/grbl.wiring.docx)

You have mentioned that your Limit pins are working but are they outputting the correct state IE Normally Open or Normally Closed.  The homing process is to clear Z axis from work area and then move X & Y simultaneously to their home locations.  if homing is selected and switch is closed when it should be open (for example) then the GRBL will alarm straight away as it cant move as it will appear its already up against the stop.

Hope this helps, let us know how you get on.

Napalm1432, 2018-12-11:
So while waiting for a response, I've also been looking trough the code, to see when an error is thrown.
From what I can see @MeJasonT is on point with the $27, the code will throw an error when switches are still activated after pull-off.
I tested all switches thoroughly, they are in the NO state, ? command confirms all that, also when manually pressing and using ? command it's confirmed. although i have to say, i did have some issues at first, but i was able to resolve that quickly after finding out it was actually attached to spindle pin, so for now I simply took out the #define for it and made sure the pin was correct.
When it comes to the position of the endstops, I didn't really think about the direction, BUT I actually make X Y home to the back right corner (seen from the front), and Z to the top, guess I'm lucky with that.
I'm not near the machine right now, when I am I will try to set the pull-off distance to 4mm to see if that solves it!
I'll edit this comment, when I have tested.

MeJasonT, 2018-12-11:
Ok Buddy, I will catch up with you tomorrow.

Napalm1432, 2018-12-12:
Well, that did it!
4-5mm is a bit much, but at least it works, time for tweaking :)
Found out my y homing direction was wrong, just had to set $23 to 0
Thanks for the kind help!

closed 2018-12-12 by Napalm1432