grbl/grbl #1480: Homing issue

oliv690, 2018-12-20:
Hello,
frist, sorry for my bad english. I will try to be clear as much as I can.

I'm new with GRBL and Arduino.
I modified a K40 laser from the original card to an Arduino card.
I installed GRBL 1.1g

I'm trying to use GRBL and sending good parameters but I think that I made a mistake somewhere because the homing is not working as I want.

Firstly, homing on Y axis was not working.
I checked switch. It's an optical switch.
When I check it between signal and ground, I have 1.1V when homing is done and 0V when not on homing position.

I modified the config.h to have homing on X and Y simultaneously. Z is disable.

When I'm trying to have an homing, if I wire motors and switches in "normal" position, the homing is done but I have the message "homing fail".
If I wire motors in "revert" position (X instead Y and Y instead X), the homing is done well, without message.

I try to resolve this issue for a long time (3 weeks) but I can't find a solution. 
I checked the wiki and I read all explanations I found concerning GRBL and $x parameters but I can't understand why it's OK in revert wiring and not OK in direct wiring.

I used $3=2 to revert Y axis.

I hope that I have been clear? (I'm not really sure ...)

Thank you for your help.

Here under a copy of the GRBL configuration.

$0 | Step pulse time | 10
-- | -- | --
$1 | Step idle delay | 25
$2 | Step pulse invert | 0
$3 | Step direction invert | 2
$4 | Invert step enable pin | 0
$5 | Invert limit pins | 1
$6 | Invert probe pin | 0
$10 | Status report options | 1
$11 | Junction deviation | 0,010
$12 | Arc tolerance | 0,002
$13 | Report in inches | 0
$20 | Soft limits enable | 0
$21 | Hard limits enable | 0
$22 | Homing cycle enable | 1
$23 | Homing direction invert | 1
$24 | Homing locate feed rate | 100,000
$25 | Homing search seek rate | 1100,000
$26 | Homing switch debounce delay | 250
$27 | Homing switch pull-off distance | 2,000
$30 | Maximum spindle speed | 255
$31 | Minimum spindle speed | 10
$32 | Laser-mode enable | 1
$100 | X-axis travel resolution | 158,000
$101 | Y-axis travel resolution | 158,000
$102 | Z-axis travel resolution | 10,000
$110 | X-axis maximum rate | 10000,000
$111 | Y-axis maximum rate | 10000,000
$112 | Z-axis maximum rate | 10,000
$120 | X-axis acceleration | 190,000
$121 | Y-axis acceleration | 190,000
$122 | Z-axis acceleration | 10,000
$130 | X-axis maximum travel | 300,000
$131 | Y-axis maximum travel | 200,000
$132 | Z-axis maximum travel | 10,000

oliv690, 2018-12-26:
Hello,
I see that there is no answer on my request. 
Is it because I'm not clear on it?

Or is it because nobody know what could be happen with this issue?

oliv690, 2018-12-26:
Yes, I'm french (as you can also understand by my poor english).

Why you think that my feed rates are creazy?

I checked again and and re-wired switches.
I have 5v when "outside" and 0v when trigerred. (now both on X and Y axis).

When I request the homing, X axis seems OK.
Y axis arrive in homing position and I can ear 6 or 7 "clac" before having the homing fail.

chamnit, 2018-12-26:
Your feed rates are fine. He's referring to the use of commas, rather than decimal points. Just keep an eye out on stability. You are near the upper end of the maximum 30kHz step rate supported by Grbl.

When you get a homing fail, do you have the error code? Also when you say "clap", do you mean that your steppers are losing steps? It's likely you have the coordinate system reversed and/or the homing direction inverted.

oliv690, 2018-12-26:
OK, thank you chamnit for the explanation concerning the comment of jahnj0584 concerning feed rates.

When I have the homing fail, when I put the mouse on the message, the box indicated the error. 
I assume that it's the error 8 in the GRBL documentation (Alarm code Description).

I tried already to increase pul-off setting from 2mm to 4mm but no change.

I also checked many time wiring but no change too.

oliv690, 2018-12-26:
When I say "clac", it's not same as when motor is loosing step.
I think that it's more like motor is trying to find the good position by going 1 step and go back 1 tep.
It doing this about 10 or 15 times and stop with th message "homing fail".

oliv690, 2019-01-22:
Hello,
I'm coming back again with the same problem.
I changed the Arduino card, the shield, the limit switches (I use now mechanical instead of optical), I add 100 nf capacitor in parallel with switches, I tried grgbl1.1.f instead of 1.1.g, , I changed pololus, I tried GrblPanel, laserGrbl and Arduino IDE, I checked all GRBL Parameters and I tried to change some, I changed $27 for 2 to 1, to 10, to 15mm, I tried to change $ 24, 25 and 26 parameters BUT nothing helped me to resolve this issue.

I explain again what happen (small changes from last time):

Sometime, I don't know why, homing is working (sometime on X, sometime on Y, but never on the 2 axis at the same time, and very difficult to re-do wit success another time. If I try to re-do the homing, without changing anything, it's no more working.

When I sent $hY or $HX (I modifed config.h to be able to try homing on axes separately), the axe X (or Y) move up to touching the limit.
It stop, return back from a very very small distance and try to find again the limit.
It don't find and then it return back "alarm 8" or "alarm 9" - it depend).
If I pish the switch manually before returning the alarm, the homing cycle is completed with no error.

I really don't know how to solve this issue.
I try to solve it for 3 months now and I really don't understand what happen.

Have you an idea of what and where could be the problem and what solution would be possible?