grbl/grbl #1483: Errors on z-axis ONLY

jdargot, 2018-12-23:
Hello everyone,

I have encountered probably the weirdest error yet when it comes to CNC firmware. 

I built a shapeoko knockoff running GRBL firmware on a mega and ramps 1.4. The particular program I am using has worked consistently in the past off of the classic Universal Gcode Sender. Recently, I have tried running this off of the classic as well as the 2.0 Nightlybuild with the same results in both cases.

Commands work for anything in the x and y directions but it gives me a load of errors any time I jog the z-axis more than 1mm at a time or any time I run gcode that involves z motion. Usually, the result is minor jittering from the other motors (as in the program actually reports movement and wild coordinates in these directions) and errors such as "Unsupported Command" and "Error while Processing Response". The ladder often repeats itself until I shut down the program. 

I have tried swapping the z-axis driver, updating the board, and running on different computers. It still has the same issues and I can't seem to find any examples online with a similar situation. 

Here are my current settings in case it helps:

$0=30 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.050 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=160.550 (x, step/mm)
$101=160.550 (y, step/mm)
$102=7524.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=25.000 (x accel, mm/sec^2)
$121=25.000 (y accel, mm/sec^2)
$122=25.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=325.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

And here is an example of the error:

>>> G21G91G0Z1F10
>>> G90 G21
ok
ok
>>> G21G91G0Z-1F10
>>> G90 G21
ok
ok
>>> G21G91G0Z2F10
>>> G90 G21
ok
ok

Basically: up one mm: ok. Down one mm: ok. Up two mm: tries to kill itself... Literally, instead of moving the z-axis, it either stops or overruns its boundaries in the x and y directions. Sometimes it gives a legitimate error and sometimes I just get the response above. But usually, once that happens, no other commands will work until I disconnect.

Has anyone had a similar problem? Are the wild stepper commands to other motors indicative of any particular problems? 

Any advice is appreciated. 
Thanks.

jdargot, 2018-12-23:
Here is the response I got after making some edits to the homing. I didn't know GRBL used such fowl language... Haha.

[image]

jdargot, 2018-12-23:
[image]

chamnit, 2018-12-23:
Your z step per mm setting is extremely high. With your max rate, you are exceeding Grbls maximum 30kHz step rate limit.

closed 2018-12-23 by chamnit

jdargot, 2018-12-23:
You are a life saver! That fixed it, just keeping the steps/mm lower.

That brings up another question though. Do you know how to make edits so distance calibration is still accurate while still meeting the steps per mm standard? Or do I just need a bigger lead screw? 

Thank you so much.

langwadt, 2018-12-23:
unless you can reduce any microstepping of you driver, you have to keep the steps/mm, instead reduce the z max rate ($112)