grbl/grbl #1486: Machine moving slow when setting up step/mm to real value

fatmisami, 2018-12-29:
I made a CNC machine myself. 
Electronics : Arduino Uno + CNC Shield v3 + 4 A4988  moving 4 NEMA17.
The NEMA 17 are moving my axes with belts and pulleys. When mesured, each mm is 5 steps.

With $100=250 , my motors are working just fine. but once is set it to $100=5 , the machine becomes reeally slow ! Motors are vibrating and everything acting weirdly. 

Any clue ?

109JB, 2018-12-29:
What kind of belts and pulleys? Number of teeth on pulley, pitch of belt teeth, etc.  What microstepping are you using?

Generally on a belt driven machine you want the smallest diameter pulley you can get, and you also want microstepping up high enough to get a good number of step pulses per mm.

fatmisami, 2018-12-29:
I am using the GT2 belt. 2mm pitch, 6mm wide [https://cdn-shop.adafruit.com/datasheets/gt2tooth.jpg](url)
Along with these 20 teeth pulleys [https://sc02.alicdn.com/kf/HTB1XM5BdlUSMeJjy1zk761WmpXap/227842221/HTB1XM5BdlUSMeJjy1zk761WmpXap.png](url)

I have set the 2 X axis motors on 16th microstepping, problem is still the same, i thing i will try 32 ..

Norwegian-Gadgetman, 2018-12-29:
You have 5 steps per mm?
I assume you mean 5 FULL STEPS per millimeter?
(200 steps divided by 40mm on the pulley, gets 5 steps)
Please note that $100 needs to be adjusted for stepping level, 
With 16x microstepping the number should be 80, with 32x it should be 160.   
Or rather, it should be in that area. The exact number can only be found by trial and error, lots of error...   
(Moving set distances and measuring the real distance moved, adjusting variables, trying again, and repeat...  )

fatmisami, 2018-12-29:
Oh thank god! 
I was seeing everyone having numbers like at least 70 there, so i thought something was wrong with my machine, can't have a 5 out there.

I will work on it tomorrow and keep you guys updated! 
Tack!

fatmisami, 2018-12-30:
UPDATE : Still not solved.

So i linked the pins on the CNC shield like this [https://cdn.instructables.com/FFR/W25T/IWQV78FT/FFRW25TIWQV78FT.LARGE.jpg?auto=webp&crop=3:2](url)

I did the 16th microsteps, but my motors are still acting exactly the same, they still need 200 steps for a revolution. 

Only 1 motor is 1/4 for a reason i ignore .. even if all of them are wired the same way.. 

Question : should i solder the cables in there ?

jahnj0584, 2018-12-30:
Just use this
https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#MotorStuffPrecise

fatmisami, 2018-12-30:
That's what i did.
I was on 200 steps/mm 
But 200 steps moved my machine 40mm

So i set it back to 5steps/mm, then the machine starts moving slow like hell !

jahnj0584, 2018-12-30:
So change your feed rate and Max axis rate

Norwegian-Gadgetman, 2018-12-30:
> 
> 
> That's what i did.
> I was on 200 steps/mm
> But 200 steps moved my machine 40mm
> 
> So i set it back to 5steps/mm, then the machine starts moving slow like hell !

How do you 'move it 200 steps'?   
G0 X200   doesn't mean '200 steps' it means  200mm
(Assuming you're running it with mm, not inches of course.)

fatmisami, 2018-12-30:
I moved my machine 1mm ( means 200 steps ) and it actually moved 40mm.

So i set the grbl to 5steps/mm.

And now that i am looking at it, thr GRBL settings are in steps and mm, means if you go from 200 steps/mm to 5steps/mm and you still want to go with the same speed, you need to go give that max rate a 40x multiplier. 

It would have been much simpler if everything was in steps i think. Steps/mm , steps/min and steps/sec2 .. because now if you change setting you need to take out thr calculator and figure out everything again .. 

I will check this and tell you what's up.

fatmisami, 2018-12-30:
Problem solved. 

It was me, i thought setting up the grbl needed only changing the steps/mm, but once you touch those all your settings are broken, especially when going from 200 to 5 ..

rabban, 2019-11-01:
What did you do how did you settings end up looking. I have the same problem

fatmisami, 2019-11-01:
Some of the poles were not wired properly, i just had to solder them.

JaydeepD-234, 2020-09-27:
@fatmisami........I am  using Arduino Uno + CNC Shield v3, 2 nema 17 motors with 2 A4988 drivers. My X axis is attached with M8 x 1.25 Pitch rod & y axis with GT2 timing pulley & 2 mm pitch belt. I've tried 8th & 16th microstepping. But I can't sync the X & Y axis at same feed rate. The y axis with belt has same problem as yours moving very slow. please help me with this.

jahnj0584, 2020-09-27:
You have to calibrate the steps/mm for each axis first. Then the speeds
will be the same.

fatmisami, 2020-09-27:
I haven't done anything with that CNC machine for over a year now.
But i remember that i did wire the pins properly ( the ones that needed to be connected to chang the 1/4 to 1/8 or so ), then i went again through my grbl settings and it was solved. Try doing the same and send in here your grbl settings so people can look into them.

ragulcool3, 2020-12-12:
> @fatmisami........I am using Arduino Uno + CNC Shield v3, 2 nema 17 motors with 2 A4988 drivers. My X axis is attached with M8 x 1.25 Pitch rod & y axis with GT2 timing pulley & 2 mm pitch belt. [...]

I too had the same issue.. just remove the shield from Arduino and solder the microstepping pins to 1/16, then multiply your old steps per mm by 16. It should work. Sometimes connecting the microstepping pins using jumpers is not working.

bodarhitesh, 2021-03-18:
What did you do how did you settings. I have the same problem
@fatmisami