grbl/grbl #1487: Problem with y axis only

jringling, 2019-01-02:
I have no travel on the y axis. I have established that the motor, driver, and cable are all working. I can find no specific setting that wold disable the y axis only. 

$$ shows v1.1z. I do have $80=0, $81=0 and $82=4 showing, but cannot find any description of these parameters.

 What else should I check?

jringling, 2019-01-02:
Took a chance and changed $82=0 and now I have the y axis working. I have no idea why...

109JB, 2019-01-03:
It's not a problem.  I just want you to be aware that you likely don't have the "official" version of Grbl v1.1 based on the $ settings and the version report. If you want the "official" one it is available  here: https://github.com/gnea/grbl

And we are currently in the "official" V0.9 github.

The only other "Official" version I know of at this time is the Mega fork of Grbl v1.1 for the Ardiuno Mega 2560 which can be found here: https://github.com/gnea/grbl-Mega/wiki.

There are many many other forked versions of Grbl on github, and most are not well documented if at all. Also I have bugs in some I have tried and decided they are not worth the trouble. The other things that happen with the forked versions are that they are not kept up to date with the "Official" version improvements. There are a few that are worthy, but for the most part just stick with the official versions.