grbl/grbl #1489: Backlash compesation

mkaluza, 2019-01-06:
Just to throw my two cents here.

First of all - thank you @chamnit  for making this software! It makes building amateur cnc a snap and dirt cheap! You should have a monument somewhere (cnc milled of course :D)

As for the backlash - I realize arduino is tight on memory - esp Uno. But I think the argument for implementing backlash compensation comes from dividing responsibilities: grbl, being a motion controller and closest to the machine has (or should have) the most intimate knowledge about it. This includes any imperfections... So when it is given a gcode that should draw/mill a square, it should mill a square, a circle should produce a circle and so on. Not a "circle, but...". I think the point of gcode standard (ISO) being a standard is that the same gcode should produce the same thing on different machines and the designer/cad should not care about specific machine peculiarities. I don't meant "type of machine", because here the designer/programmer should care, just a particular unit. Imagine you build two identical machines - or at least you try. They won't be perfectly identical - ever, especially if you're a hobbyst. And each of them should have a way to compensate for this so that code generated for them produces most similar result possible.

Don't get me wrong here - I'm not saying here that "I'm right and you must do it". I think it's just a low hanging fruit that can improve the accuracy of hobbysts' machines almost for free (I know that has to be written, but it's only once - then can be used by many). Because it's for hobbysts, right? hobbysts will produce imperfect designs... so why not make it more hobbyst friendly?

I know that one could always upgrade their machine, but I think that this argument is not valid - this usually costs money and time, so probably if they could spend them, they would (I'm speaking for myself here right now). Besides that's why uP are everywhere - they allow to cheaply fix/improve things with software that would otherwise be expensive/hard to do. So it's about getting the most for their money.

The same arguments apply to geometry correction - I know my machine's axes aren't exactly perpendicular and I know I did the best I could - If I could do it better, I would :)

I know some CAD/CAM can do those, but that's _some_ - it limits you to only those that can (from an already quite limited subset of free software).

Now for some solutions.
I'm a programmer, I could help - I'm  almost done building my machine and as soon as I catch up with the rest of my life that has been put away I'll be tuning it and both backlash and geometry will be an issue. I'd just need some guidance where to start. Whole code has only 7k+ lines, so it shouldn't take long to get up to speed.

As for UNO limits I have two ideas:
1) ifdefs - maybe make some features optional (either configurable or hardcoded) that would be only available on Mega boards. I know it requires careful coding, but it's an option
2) If you say "upgrade your machine", I'd say "upgrade your uP" :) Arduino Mega clone from Aliexpress is $8, uno is about $3... that's probably one of the cheapest items in the machiine... even for a very cash limited project $5 shouldn't make that much differrence (or +$15 for the original). And that would give **8 times** more space for the code and **4 times** more ram/eeprom. Added bonus is much more I/O pins. No other single thing will give you so many "times" per dolar :D 

I'll say it again - I don't want to force anything on you, just to open a discussion.

P.S.
I totaly agree about not supporting some advanced gcode features like loops, subroutines etc. It's the responsibility of a gcode streamer (unroll loops, load code from other files). So I'm not an "I want everything" guy.

Have a good day :)

jove1, 2019-05-13:
Hello Marcin,

I've seen you've implemented backlash compensation on grbl-mega,
Which branch in your github is in the most ready state?
I'm using ramps, I've read through some of your cleanup patches and
on first read they make sense. I don't need skew axes or UVW relative moves,
though it seems only config parsing is implemented.

Thanks for recommendations

PS I don't seem to be able to contact you on mk@flex.pm (no MX record).

Jozef Vesely

bwanamukubwa, 2019-05-28:
Hello, I saw the README mentions backlash but i'm not see it used - could you please provide a few descriptions of the contribution? 

Thank you in advance.

mkaluza, 2019-05-28:
@jove1 sorry for the late reply. You can use anything you want. Edge is more stable, test might not be so (but I'm using that branch and it works without problems - sometimes I have problems when changing feed override, but this might be my machine, not the code).
So you can build either 'edge' or 'test' or use a prebuilt version in releases (it's built from latest 'test', I use exactly that binary).
Skew is implemented in both branches, see https://github.com/mkaluza/grbl-Mega/commit/562d7f8ca7bebe8b925c66aaf7191adc21fae11d

@bwanamukubwa I don't know where are you looking and what you mean by used? where? It's only in my repo - upstream refuses to use it, so I gave up the discussion.

MrMabulous, 2019-06-03:
What’s the state of this? I’m currently building a little CNC out of a proxxon MF 70, which has horrible backlash, and try to determine whether it’s better to go with grbl or marlin (grbl seems more cnc focused but marlin seems to have backlash compensation)

109JB, 2019-06-03:
I looked into using Marlin for CNC milling and I do not like Marlin for milling machines for several reasons. I do use Marlin on my 3d printer though.

First because they have haphazardly utilized G-codes that have been historically used for other purposes for their own new features.  No problem there, after all it is their program, but having been a CNC machinist/programmer I can't get past their use of g-codes that have historically been reserved for other things. I have no idea why they chose to do this as there are plenty of unused integers out there that could have been used for their purpose. G10, G28, G30, G33,G42 all have different meanings in the CNC milling world and even though Grbl doesn't support a few of these, Grbl has remained true to the historical use for codes it does use.

I don't see any tool height feature within Marlin. So if you are going to do more than 1 tool jobs it is going to suck. Unless I am wrong about this.

Marlin only has 3 realtime commands and you have to enable the "emergency parser" to use them. Basically the only one of real value to milling is the emergency stop. So, you do feed hold, resume, feed override, spindle override, rapid override, toggle mist coolant, toggle flood coolant , or toggle spindle that Grbl has.

There was more, but I can't recall at this time.

jove1, 2019-06-03:
> What’s the state of this? I’m currently building a little CNC out of a proxxon MF 70, which has horrible backlash, and try to determine whether it’s better to go with grbl or marlin (grbl seems more cnc focused but marlin seems to have backlash compensation)

I got to < 0.010mm backlash on MF70 by tightening the nuts at the end bearings ....