grbl/grbl #1490: Accurate spindle speed data instead of pwm

mmcristi1981, 2019-01-07:
I have a custom spindle controller with a separate arduino board separately from the grbl running one. I would like to read/get the current spindle speed from the grbl board. I would like to know if it is possible to get the information in a binary format (e.g. emulated serial communication) that periodically sends a two byte number that is the real programmed gcode spindle speed instead of the current pwm signal. I can try to change this in grbl myself, but maybe there is already something like this available or planned. 

The alternative of trying to read the existing pwm and convert it back to a number presents several problems: 
- lost accuracy (only 8 bit pwm signal + conversion accuracy losses)
- need to know on 2nd board the maximum spindle speed currently set on the grbl board, which can change
Another alternative could be to have both arduinos connected to the host PC and change the GUI to constantly send the current spindle speed from one board to the other. 
However I think it would be more elegant and accurate to have the boards communicate directly somehow if possible.

shooter64738, 2019-01-07:
If you put grbl on a mega 2560 board you will have multiple serials available. I use this method to send Gcode to the motion controller, and some Mcode and spindle data to my spindle controller. The coordinating mCU is connected to the pc via usb. The gcode line is processed on this coordinator, then motion code data is sent to the motion controller, spindle data is sent to the spindle controller. I use an encoder on the spindle control to determine true spindle speed, and a PID loop tries to hold the S value speed that was specified in the programmed line for the spindle.

You could do something similar by grabbing spindle data and send it to your spindle controller from grbl, on say serial1, 2, or 3. just be careful you don't tie up the processor sending data too frequently. 

It might be a cleaner foot print than trying to have 2 usb devices talking through an app on the host pc.

MeJasonT, 2019-01-07:
This might be of interest to you
[https://github.com/gnea/grbl/issues/558](url)
this was my solution

mmcristi1981, 2019-01-08:
Thank you, both are valid ideas in my opinion. I tend to lean towards the sniffing approach if nothing else works. It would be great however to have a simpler solution. I am thinking, with a configuration option and some additional GRBL code would it be possible to change the function of the pins 11 (PWM spindle pin) and A4 (not used/reserved) to act as a rudimentary RX/TX? The communication speed doesn't have to be great. Hack, I think in theory it would even work with just one wire (the spindle PWM). It could be unidirectional, and GRBL could just periodically send this (say every sec). The pulse train would need a long pulse or smth like that as a delimiter for the data bits. Sending 16 bits + delimiter signal every second shouldn't be that difficult. The receiver could take care to filter out the wrong bits, no problem if a package is lost, wait for next one, acknowledge is not needed in my opinion. The only problem is probably the available MCU resources on the GRBL board (program memory, available processing power etc).

109JB, 2019-01-08:
Why not just sniff the status report responses. The current spindle speed is right in the report:

<Idle|MPos:0.000,0.000,0.000|FS:0,**_3500_**>

MeJasonT, 2019-01-08:
Using A4 and pin 11 it is possible to send a value of between 0 and 511, where pwm = 0 to 255 
pin 11 switches on allowing the use of 0 to 255 on pin 11 and A4 is high or low adding 0 or 255 to the value.  I had looked at transmitting a frequency out of Pin 11 which would be a value of say 0 to 5khz but it gets very complicated very quickly.  You will also end up with a situation where you wish to halt the code and have to wait for commands in buffer to empty before the spindle will stop - not safe operation really. 
you do not have any pwm pins free so you will need to bit bang a pin (actually better than PWM) Outputting 2x pwm outputs and summing the output can give you 16 bit, keep it analogue don't bother with a delimiter, your 2nd arduino would require a ADC which could read 16 bit as its own ports can only read 10 bit.

[https://www.youtube.com/watch?v=yjgZ3dQ-PFw](url)

using the arduino 16bit timer to bitbang pwm may be possible but i think sonny will have utilised it to the max already.

109JB is an authority on grbl and works closely with Sonny (he also wrote a great GUI),  It really is worth taking his advice on board.
changing code in grbl is not a simple task as the memory is already fully utilised and very time conscious, how the hell they get so many high speed operations to run in the loop baffles the crap out of me.  I'm currently working on an EDM project where i'm reading analogues, displaying on LCD driving a stepper motor and outputting a pwm signal.  The LCD activated in code kills the stepper, there are no blocking functions written in the code (mills counters and no delays) - LCD is on I2C backpack though.

MeJasonT, 2019-01-08:
Try this - you will see how easy it is

[image]

open serial monitor in Arduino IDE and set baud rate to 115200 you will see the data stream shown in 109JB's message - <Idle|MPos:0.000,0.000,0.000|FS:0,3500>

the analogue output idea can be improved upon using circuitry to such a point that a second arduino possibly wont be required.

an advanced 16 bit pwm output

[image]

[https://hackaday.com/2013/01/17/making-better-noises-with-dual-pwm/](url)

MeJasonT, 2019-01-08:
mmcristi1981 thanks for your feedback.
i didn't think of using A4 to indicate hi/lo byte, don't know why, perhaps i had only just had my first coffee.
65536 RPM potentially is an unbelievable speed only achieved by proper top end industrial VMC's, do you really plan to go to that sort of speed. scary.  the direction is already output on another pin so you already have +/- rpm signal available.  I would tend to leaving the second MCU taking care of the spindle and using an additional alarm Wire from it to return to GRBL controller wired as the safety door switch or in addition to door switch if already used to tell GRBL that the spindle has crapped itself.

The sniffer method is actually quite fast almost real time within 200ms or whatever refresh rate you have your reply messaging set at,  perhaps you are worrying too much about a closed loop system, at the end of the day your 2nd MCU could tell you lies when the motor isn't even spinning - unlikely with the hall effect sensor but if you consider that a vast majority of users are using a Dremmel and sending the PWM signal to drive it with no feedback to GRBL then you have to consider the cost of your belt and braces method.  

The changes in spindle speed tend to be in between code blocks so ramping of rpm up/down is not much of an issue - if you are operating at rapid speeds then add a delay/wait into your post processing when you change spindle speeds.  The spindle speed will be constant throughout the operation only feedrate is reduced for cornering and trochoidal milling etc

Start making chips its much more fun.  I'm slowly starting to release the beast, I have been scared by the speeds recommended for milling Aluminium etc but have finally grown a pair.  Not got a steady stream of chips flying out from the tool yet but almost unfortunately i only have 7500 RPM to play with on an old Elliot milling machine i converted to CNC.

mmcristi1981, 2019-01-08:
No, lol, I do not plan to go that high. I was just trying to say that 16 bit covers pretty much every scenario you could think of, apart from controlling maybe a small gas turbine engine with grbl. Has anybody considered using that as a high speed spindle ;)
Agreed, making chips is fun, but when the spindle slows down due to load controlled only by pwm, fun turns into frustration pretty quickly.
I plan to peek into grbl code and see if I understand anything. We'll go from there.

109JB, 2019-01-08:
if you sniff the status reports, the longest period you would lag in commanded speed is the timing between status reports. I generally set this to 0.1 seconds between status requests (?).  It seems to me that is faster than a spindle could actually change speed anyway.  If you need more assurance the spindle is at speed before proceeding to the next command, you could program a small pause into the g-code that grbl is receiving.  Something like:

(Spindle already at 3000 rpm)
M3 S4000 (Increase speed to 4000rpm)
G4 P0.2    (0.2 second delay to allow the spindle to get to the new speed)
(Other commands here)

Or rather than modifying your g-code for this, you could modify or write a GUI to automatically send the G4 Px command after every M3 or M4.

Or if you want better, you could write or modify a GUI to intercept any M3 or M4 commands and wait for feedback from the spindle controller before sending more commands.

109JB, 2019-01-08:
@MeJasonT - Thanks for the plug.  I don't actually work on the Grbl code itself much. I'm only an amateur programmer with Visual Studio, but I'm a former machinist that ran some, at the time, stat of the art machining centers. 

I'm glad someone out there has been able to make use of my GUI. My original GUI was written for V0.9, and I've been working on a re-vamp for V1.1 that adds many probing features, but still has the tool change, tool table, and other features I like to have. I'm simplifying the code so for several things and testing now. Its getting close and once I have it completed and do some more testing I'll get it out there for all to see/try/use. Just have to find time between other stuff to focus on it.

shooter64738, 2019-01-08:
@mmcristi1981 just an FYI, if you connect your spindle control RX line to your motion control TX line for eves dropping on the comms, you may have trouble trying to upload firmware changes to either board. Just unhook the RX line for your spindle control board when you need to update. If you plan to use bit banging or some other method, then disregard that.

mmcristi1981, 2019-01-09:
I just realized, we're probably on the old grbl issues tracker. Noticed there's also this other one: [https://github.com/gnea/grbl/issues](url)
Any idea where we should open issues? I think the other one is newer more active? I am surprised how active this one still is.
Which GUI are you discussing? You mention visual studio, that's for GUI dev right? If I want to change GRBL I understand I can do it with Arduino IDE and compile/upload directly from there. Am I wrong?

MeJasonT, 2019-01-09:
You are correct, altering GRBL Firmware is done on the Arduino IDE
although there is a tool called visualmicro which adds the IDE into Visual Studio.  

Whilst we had 109JB's attention, I thought I would ask about the GUI (front end software)
There are a few available but each has its limitations
Candle, has enough functionality and looks good but is limited in things like lacking more work coordinates options (only G54) and has no Coolant option. Judging by responses to issues it does not appear to be maintained/supported any more
GrblGru Brilliant, but graphics are not as sweet as i like.
Universal G Code sender, umm not operator friendly from a machinists point of view - pain in the but to install on windows, still waiting for a point and click install package.
Grbl Panel, looks and feels like a machine panel, like using this on a lot but unfortunately is no longer going to be supported and is lacking a few features still.
and 109JB's User interface- you just know he was a machinist - looks and feels real, tool library is a nice touch and probing is an unbelievable tool for hobbyists, costs thousands in the real world to get CMM.

Not a complete list by any means but a reasonable selection

As GRBL has limitations on the number of commands it can handle, it would be nice to see one of the control applications written with conversational program code written to cover operations such as drilling and hole matrices, its the GUI which should handle these functions anyways.

mmcristi1981, 2019-01-09:
So I had a look at the code. It looks like it could work to use A4 and 11. The logical problem I see is how to periodically send the two bytes. It is very easy to send it once when GRBL currently sets the PWM. Instead of setting one PWM in the function, I would first set A4 to 0, set first byte PWM, wait a few ms to give the chance to client MCU to pick the pwm train and convert it back to a byte, set A4 to 1, set second byte PWM and done. Unfortunately we will get no feedback that the client received the bytes correctly. So  best would be to repeat this in a loop even if the speed doesn't change. I could do that in the report call, but as a concept I think this is hacky. I would prefer to do this periodically even if the GUI doesn't ask for a position report if spindle speed is in this special config option.
It would be great to maybe get some imput directly from mr. @chamnit how to best achieve this periodic update. I don't know if there is already a routine that is periodically called by timer interrupt or something where we can afford to inject the spindle update code. The problem is this code either waits the few millis between byte sends when the spindle_sync function is called, or I write it without the delay but the code is called often enough that I can continually cycle the sent bytes. Cycle 1 (timer interrupt 1), send MSB, Cycle 2 (timer int 2) send LSB, Cycle3 send MSB, C4 send LSB, I think you get the idea. Problem is I don't know if GRBL has such a cycle we could use.

shooter64738, 2019-01-09:
@mmcristi1981 I have posed questions on the 'newer' grbl hubs and got little or no responses. I think even though this is the 'older' one, it gets the most traffic. 
Not sure where/how you could add a pwm,delay,pwm configuration in there. Its constantly looping in protocol.c but its also adding to the st_buffer as the motion is running. Anything you add in there that slows it down will have some ill effects on motion drive.
If you tried sending 2 bytes out on the serial, that might cause the gui to behave strangely because it would assume its a comm from grbl. Reading the PWM output from grbl might not be as bad as you think. It should be a pretty steady value. 
Timer0 and timer1 are occupied as the stepper driver. Im not sure if timer2 has a hardware pwm pin thats available but if it does, you could just set that timer up to send the most sig byte on the first loop, then change the delay on the timer on the second loop, and it would send the least sig byte. Since that's handled in hardware it might not burden processing much at all. It would repeat those 2 bytes over and over. But that might send the msb multiple times, then send the lsb multiple times. Would have to turn the timer off at the end of the byte.   Just kind of thinking out loud I guess... 
I work with the 2560 boards, so I'm not 100% sure which timers might be available on the 328 boards. 

@MeJasonT You could go grab Atmel Studio version 7.0 from MicroChip. It IS Visual Studio but wrapped up with their own keywords specific to writing asm/c/c++ for atmel chips. Comes with a handy chip selector when you start a project. By the looks of it I would say its based on Visual Studio 2010 or 2013, But its nice to use for the atmel chips, best of all its totally free. It also nice to write the code in visual studio compile it and run through the code on windows with the debugger, and once the guts of the code is working, you can implement it in atmel using hardware abstraction. You can even have a solution with multiple projects and reference them just like visual studio does. 

I believe 109JB 's gui does do canned cycle conversion already. I think there is one other UI that does it. I have implemented that type of work on the hardware so it appears that you sent a canned cycle, but behind the scenes it just splits it up into multiple motions, and sent one at a time. My 'GUI' is a piece of aluminum with buttons and switches. Which I seem to be spinning my wheels to finish.

MeJasonT, 2019-01-09:
Hi shooter64738 thanks for you reply.
Do you have a link to your GUI i would be interested to see it.
on a side note - do you know of any links which show how to break down commands to basic G commands/operations, I know that in cambam (my CAM application) that someone wrote a script to produce drill functions that will run on grbl to replace grbls missing drilling functions.
[https://www.reddit.com/r/shapeoko/comments/5pvsy2/cambam_drill_macro_to_replace_g81_canned_drill/](url)

Unfortunately cambam is void of a plane operation, for what reason i have no idea.  there is a workaround which isn't too much hassle but it would be nice to just click an operation and then click on a face.

shooter64738, 2019-01-10:
@MeJasonT I dont mind sharing it. I intend for it to be open source. I had it on GitHub but something happened to my repo and it would no longer upload so I deleted it and will be uploading it again.

shooter64738, 2019-01-10:
Here is a link to the repo: [Talos-CNC](https://github.com/shooter64738/Talos-CNC).
Its a large project, consisting of a pc client and a raspberry pi client for the 'UI' which I have made almost no progress on because I dont really want to use a pc. I prefer a hardware panel with buttons and switches. There is an atmel solution for both the AVR and the ARM atmel chips. One of the libraries is support for a 4x20 lcd display. I do not have my spindle controller integrated into the solution yet, but I'll work on it when I have a time. 

The canned cycles, and cutter radius comp mostly work, but my machine is undergoing a rebuild and I have not been able to run the firmware on an actual machine for some time. I have been reduced to running the firmware in simulations in windows and looking at the output motion commands. There are instructions for how to compile the firmware to run in windows up there if that interests you.

109JB, 2019-01-10:
Hi guys.  I'm only able to get on here sporadically, but I saw the most recent posts about GUI's.  I don't want to say too much since we're kind of thread-jacking, but did want to answer 2 things about my GUI.

1. Canned cycles.  In my old GUI and my new one too I have some of the canned cycles implemented. (G81, G82, G83, G85, G89). I can't remember how the old GUI was but in the new one you can use these cycles in MDI mode or put them in a g-code program. They fully support G98, G99, G90, G91.  The new gui also allows you to do a cycle on one line input, and follow the next line with just X,Y cooed to repeat the cycle. So,

G81 X1 Y2 Z-1 R0.1 F3  
X4 Y5
X7 Y8

results in 3 drilled holes 1 inch deep at (1, 2) (4, 5) and (7, 8).  You use G80 or any other motion mode (g1, g2, g3, g38.x) to cancel out of the canned cycle just like on an industrial machine.

2. Cutter comp.  I played around with getting cutter comp working and had a tester that worked for straight line moves. The problem I had was when you throw curves into it and resolving the new intersections of the offset path. I may play around with it more sometime, but for now it is only in one of my test program as an experiment. For how I use my machine I don't often need it anyway, and when I do I just work around it. It is usually when I'm milling a bearing pocket or something like that where it would be useful, and I generally just use another work offset, center on the bore and then manually edit code to creep up on the fit I want.

MeJasonT, 2019-01-10:
Its a shame Git don't put a forum tab at the top of the page so we can take the conversations away from the issue posts.  I found using Git as a file repository a pain from day one, you need a degree in programming to set the damn thing up and when you changed files it had a habit of automatically saving the current document state even when you didn't want to. Ie just experimenting with a bit of code in your program which you had no intention of retaining.  Ok you could back out of it but why save it in the first place.

mmcristi1981 appolegies for hijacking your post, things just took on a life of their own and seemed relevant to a certain extent.  If we are all happy perhaps we should consider moving to a new fork.
how do we do that John, is it just a case of opening a new fork called GUI discussion perhaps.

shooter64738, 2019-01-10:
@MeJasonT Generally, you would open a new issue related to your topic. You could just open an issue in your github repo, or mine. 

@109JB I spent an awful long time on cutter comp. Arcs do complicate things. Here is my write up on it, if you find some use from it. [CRC](https://github.com/shooter64738/CRC/wiki)

@mmcristi1981 I did look for an extra timer on the 328P. I dont see one thats available. The one I was thinking you could use, is already being used for the spindle PWM. Bit banging off of a digital pin would be the only other thing I can think of and timing that might be difficult. Best of luck to you, and I apologize for the thread jack.

mmcristi1981, 2019-01-11:
Don't worry about hijacking. I am also interested GUIs. But it would be best probably for other users to have the discussion under a name that attracts attention to the intended subject. Somebody looking for GUI info will most likely not look into this issue because of the name.
So I've been thinking more about a solution to my problem, as I didn't have time for anything else but think about it. 
Because of the problem of repeating the byte send, I think it would be a better approach to use the existing PWM singnal 11 as an external clock carrier and at the same time as an internal interrupt source. In that way I could send the information only on pin A4, calculated and clocked by the pwm. Receiver would have an interrupt based on the pwm signal, and read the current value from A4. I would need a bit train that signals the beginning of the package. I can split the info in 8 bit chunks then the indicator of the byte sequence would be 1111 1111 1 (9 bits), and the rest of the bytes all have to start with 0 followed by the useful 8 bits, loosing one bit per byte.  On the receiver end I would ignore the first bit which is just a delimiter. This way I can send multiple bytes, maybe including then a checksum. In this way I could even send the direction in a special package, thus only using pin 11 and spindle direction, just as many pins as currently used for spindle control.

To do this however I am missing some small but important details about how MCU's work with interrupts, as my experience with them is not much. I would be grateful to get some input on this. What I don't know:
1. What happens if I am in an ISR that didn't disable interrupts at beginning and another interrupt comes from a different source? Is it suspended and continues where it left of after the second int is handled? This means there has to be a stack, how deep is it? Also is there a priority option that can be set?
2. What happens if I am in an ISR that did disable interrupts at beginning and another interrupt comes, is it postponed, or do I lose it completely?
3. Does the time the ISR takes to finish influence the timing of the timer based interrupt that triggered it? In other words does the timer continue 'ticking' while the ISR is executed?

All this is very important to not affect the stepper pulse trains, to avoid shorter or even lost steps.

Thanks a lot for your help.

MeJasonT, 2019-01-11:
Sounds interesting, 
The voice in my head is screaming at me to say just use both pins to output values when summed represent a spindle speed and forget about checksum, delimiter and crc checks etc just send the speed and treat it as such then operation is simpler and easier to produce.  Its not my project so i will tell the voice in my head to put a sock in it.  be careful with the analogue pins, On the 328, 2 pins cant be reassigned they are input only (cant remember which).  Im sure pins A4 & A5 are fine otherwise I2C wouldn't be possible on those pins when used in those sort of projects.  Don't use I2C for this project 200ms is damn slow and will give you a headache when it delays your GRBL performance with a huge delay.  Plus its blocking, i think.  being a person with a lazy disposition i would make life simple  and not get too tied up with the error checking and feedback confirmation stuff etc.  You will discover you are no worse off than any other machine being run using grbl.  experimenting with outputting pwm on pin 11 and using A4 for bit banging the high byte you will achieve a steady stream of values to control your spindle speed.  After all the time spent trying to create a virtual comms link with the spindle controller you would still need to write code and embed the responses into grbl (keeping in mind its fully loaded already) - you could still have noise destroy the comms between your grbl and spindle.  That happens to the best of us just using the pwm output.  If you have a couple of Arduino's lying around, load one with GRBL it will run and work perfectly well as if its your machine (don't bother changing machine settings on it as it wont have limit switches and will be unable to use hardware limits or homing etc).  By default it will allow you to run cut files (job's) and treat the second as your spindle controller.  If you can get a value from the first arduino to the second using pins 11 & A4 then you will be pleasantly surprised by the speed of response and its usefulness.  If you want it to be more closed loop than that then alter the code once you know you can get the basics working.  That's probably telling you how to suck a huge egg.  But sometimes its nice to get someone who breaks the idea in your head down into byte sizes, i tend to get stuck in the complexities of my own ideas sometimes.

Did the 1 wire option not look feasible, I imagine the sketch for it will be a program in itself and probably blocking as well.  I know the temperature sensors are capable of 2 way comms. Ie you ask device at address for its temperature, you can tell it to respond in C of F and read its memory - it obviously responds with the values asked for.  there is a delay as it can only do one direction at a time.  It will suffer from high frequency noise from a spindle so would need to be wired with cat5/twisted pair.

mmcristi1981, 2019-01-11:
I do not remember about the idea using just one signal. Not sure I understood correctly.

Anyway the problem with the initial idea with A4 and 11 to use A4 low/high to indicate the byte, has a fundamental problem with repeating the send. I do not know how to solve it easily. Using the pwm as both a carrier clock signal and at the same time as an interrupt to achieve the periodic send on the GRBL board seemed like a sensible alternative. CRC was just something I thought about when I realized that with 2nd approach I could send more bytes than what I needed initially.

I already have a 2nd arduino acting as the controller of the spindle box. This will have multiple things to take care of, but it could also listen for signals from the GRBL board.

This communication thing is at the level of concept atm.

MeJasonT, 2019-01-11:
Ohh thinking about it, that's dangerous. lol.
you are at the exact stage i thought you were.

The problem you are trying to overcome as i understand it is to get better resolution of spindle speed out of grbl.  255 steps really is pretty pants over an rpm range of 0-10,000.
If this sounds like the problem and working purely on a conceptual basis just for the sake of us trying to come up with a quick fix to GRBLs limitations. then we (royal we) could consider leaving the pwm as is on pin 11, so somewhere in code (sounds like you are more advanced at programming than me) we get the value of spindle speed and divide it into a 16 bit value will give us a value of 65k, lets call that our rpm 1 bit = 1 rpm, having a machine hunt +/- 5 rpm will be insignificant at 8000 rpm just for example.

The direction pin is still not interfered with so we can still use it to provide direction information.  So a way is needed of transmitting 2 bytes using pin 11 which is already coded and working (in principle - in reality you don't get a true 0 to 255 pwm i have often found it a bit inaccurate) so lets keep pin 11 for the low byte. now comes the biggy, how to get the high out put transmitted on A4 to represent an 8 bit value.  if whatever code bit bangs pin 11 uses the split value (now 16 bit) to send Hi out on pin 4 and low out on pin 11,  using something like the tone library you could receive 2 pulsed inputs and merge them to give you the 16 bit representative value. or using the hardware option i described earlier sum the two pwm values to give us a 5v analogue value which is derived from 16 bit.  bit pain in the but is the Arduinos analogue i/p only has 10 bit resolution so an external ADC with 16 bit resolution would be required on device/arduino 2.   They call the output from pin 11 pwm but in reality it is not true pwm its a bit banged frequency output (pretty much the same thing but not quite).  good news for us is its already bit banged not tied up in a default and regimented processor command.
so telling a second pin to bit bang the high byte in the same code location using a similar command should be relatively straight forward and just happen in the code at the correct point not causing interference with the rest of the codes operation (small delay of microseconds depending on time taken for processor to handle instruction).  

Don't be too concerned about timing and syncing the 2 bytes, it will happen.  your other concern being how do you know when its transmitting the speed,  Grbl outputs the value on pin 11 when its activated and alters the speed with the S****** G code command when instructed, all off it goes dead, this behaviour wont change so generating two bytes and outputting them following some sort of summing will produce a value that we can work with.  Personally I like analogue voltages or string commands, frequencies at a push.  If you used the hardware option of summing and fed it into an amplifier (10v supply) then you would have an industrially recognised standard of 0-10v for feeding directly into a VFD.

Incidentally what signal drives your spindle, this could all be immaterial if you said RS422 or something Bizarre.  All dependant on what the 2nd arduino is doing for you.

My ideas is probably a load of pants and i will be intrigued to find out how you resolve this problem.
Everyone has a one track mind and thinks their way is the most obvious, but it is so easy to delude ourselves.  I don't have a single second to play at the moment but i'm itching to set up a test rig and have a play.

shooter64738, 2019-01-11:
@mmcristi1981 To answer some of your questions regarding interrupts.
First keep in mind not all mCU's work this way, so Im only addressing the 328p/2560 (Arduino Uno/Mega). The details of how other mCU's work could be different.
The 'stack' on these is basically 1. 
If an interrupt service routine is in your code and the bit flag for that isr is set, when those conditions are met the interrupt with 'fire' or 'run'. When an interrupt is needed a global flag is set in the mCU indicating an interrupt needs to run, the mCU will save off where it is in the main program, so it can service that interrupt and then return to where it was in the main program. As soon as the interrupt starts the global interrupt flag is cleared (but it can be set again).

During execution of the interrupt, other interrupts are 'disabled'. in other words, an interrupt cannot be fired while within another interrupt, however the global interrupt flag can still be set. so while you are within one interrupt, another interrupt can occur, but it will not stop your current interrupt and jump to the next one. It will finish the current one, and then move to the next one. If you spend too much time in that first interrupt, and the flag has been set for another interrupt, it wont be cleared until you go to the next interrupt. If a 3rd interrupt needs to happen, its going to be lost. 

It wont stack interrupts any deeper than that. If the global interrupt flag is set, it must be serviced before the next interrupt is needed. If the flag is already set and you are in an interrupt thats taking to long, any further interrupts will be gone.

So lets say you are looking at grbl and in the stepper.c isr 'TIMER1_COMPA_vect' you see some code that drives the stepper. At the top of that code, interrupts are disabled, and the global interrupt flag is cleared, if another interrupt happens while in here the global flag will get set again. (that interrupt will fire AFTER we get done with this one) If you look a little further in that code you see 'sei()' that is re-enabling the interrupts. That normally happens  by its self when an interrupt finishes, but its turned back on there manually so that the stepper reset timer can fire sooner. If you look at the code below that 'sei()'; its doing some buffering and stepper stepping. Now you think man I could add axis steps in here. Which you could, and I have. But lets assume you added 15 more axis's to step (crazy amount of axis's ) and now this isr takes longer than was intended. The stepper reset will fire just like it was supposed to, but since interrupts were turned back on with sei() and the time lapsed for the stepper reset to fire the global interrupt flag is set already. We took so long in the driver isr that it was also time to move another step, but now we are going to lose that step, because the flag was already set, and its only going to be able to fire the reset timer. So now our step timing is slowed down, and if we keep piling things on in there it will eventually stop even getting back tot he main program where it was when the isr first started this thing off. 

I've read over this and edited a couple of times and it still might not be clear.. But if its not clear to you I'll try it again. :)

mmcristi1981, 2019-01-11:
Oh man, thanks, I'll try to understand it. I've also been doing a little more research about interrupts, and I then removed the questions thinking those are not yet relevant, shouldn't worry about it atm. But you must have picked the post before the edit. So I put back the questions to avoid confusion.

Thanks.

MeJasonT, 2019-01-11:
Hey Shooter that's both barrels.  I think the moto here is dragons beware.

shooter64738, 2019-01-11:
Well its not as complicated as I made it sound I dont think. 

Think of it like a conveyor belt. If you are standing there you can grab the box off the conveyor. If another box (second isr is needed) comes by while you are away with the first box, its going to carry on to the end and fall off. (a missed interrupt) if you set the global flag (by calling sei) during the interrupt, you can be servicing one interrupt and the next interrupt will get flagged to run, but thats as far as you can go. That first one has to be done and you have to be working on the next isr before a 3rd comes in our its lost. If you do not make the sei call in the first isr, then you wont even be aware of the 2nd interrupt. Generally you dont re-enable interrupts while within another interrupt, so if the isr takes too long, any other interrupts that needed to run are not going to.

I ran into this while developing my spindle control for rigid tapping. Using a hi res encoder spinning at 10,000 rpm I would have to service an isr every 1.5 micro-seconds. Worked great, so I thought.. I was sending rpm, shaft angle, load, all kinds of things.. and then I got above 500 rom and serial slowed down.. and then I got to 3500-4000 and all it had time to do was service the isr.. Which does no good if I cant tell the motion controller the spindle speed. So I learned form that, that isr timing can get pretty darn tricky. 

This is why I was thinking, if you wanted to approach it the way you are thinking, that a hardware pwm would be the easiest. Because you can set the pwm value and the hardware takes care of it. theres no isr to run or service. But the 328 hardware pwm is already used to drive the spindle. You would need to switch to the mega board and run grbl on that. 
But the mega has 4 serial ports so you could easily talk to your spindle control via serial, which might make that pwm a moot point already. :)

MeJasonT, 2019-01-11:
Shooter, would you mind if i asked you to pm me, i have a question that you may be able to help me with running on the same lines for a non grbl project.  I wont hijack this thread again. J.

mmcristi1981, 2019-01-11:
I think I understood but let me summarize for validation:
- interrupts are by default disabled by any interrupt routine execution
- while you are in the middle of an ISR, you always loose other interrupts if sei is not called
- if you do enable interrupts with sei, the next ISR will be called when current finishes; if more than one arrives during that time you loose the 2nd, 3rd etc

Is this valid also for MEGA and say, DUE?

Thanks for your valuable insight and time it took for you to explain all that.

langwadt, 2019-01-11:
you only lose interrupts from the same source, other interrupts will run when you are done

shooter64738, 2019-01-11:
Yes, you have it!
This is valid for the uno and mega (328p and mega2560 processors.) The DUE operates on the SAM3X chip and I _*think*_ it operates similarly but I am not 100% certain. I am still learning the SAM hardware which is significantly different.

Atmel has some really good write ups in their data sheets. Just google 'atmel 328 datasheet', 'atmel 2650 datasheet', 'SAM3X8E datasheet'. Something in the descriptions of how it does one thing or another may spark an idea in your head to get you where you need to be. It will also help unravel the mysteries of some of the grbl code. Like timer operations and stepper pulses.

mmcristi1981, 2019-01-13:
So I did an initial test with the clock and data wires idea, using two arduinos. One is server that sends a 16 bit number, out of which 15 are useful + 16 1's which indicates the bit train start. Client uses interrupts to pick up the bits. Tested and works great. Haven't noticed lost numbers, but I haven't tried with longer wires and higher speeds yet. You can have a look at the current working prototype if you're interested: [https://github.com/mmcristi1981/arduino_uni16com](url) 
The mentioned server code is just to test the concept, this is not how GRBL would send the bits. 
So next step is to adapt and integrate the server logic in GRBL and make it configurable. I see two viable options:
1. Send one bit every time the GRBL main step ISR is executed by bit banging the two wires (clock and data). I need to add a mini state machine for that and make it as fast as possible to avoid slowing down the step ISR. At every step ISR execution I would advance the bit sending state machine. I am pretty sure I have enough time left in the ISR after sei is enabled. Available time increases as steppers slow down. So if there is a problem, it is only at the highest speed GRBL can work at. I could also count several step interrupts and only advance my state machine every n'th execution if sending is too fast.
2. The other option is to pursue the idea of using the available pwm timer to add another interrupt that advances the bit sending state machine in a separate ISR.

In my opinion, option 1 is better, as it doesn't add another interrupt that can happen while the stepper ISR is executed. I think this is cleaner. This also rings a bell, that maybe this idea could be used to accurately synchronize stepper movement with spindle rotation. Just an idea, might be silly. I think GRBL already does something like this for lasers, but I am too much of a noob at GRBL to be sure about that.

And I remembered today one of the main reasons why the existing GRBL PWM function is not enough for me: because I am forced to specify max speed in GRBL. Imagine I have a spindle where I can change speeds with a lever between low and high. Or you might have belt changing capability. Since I want to control the main axle speed, and that's where I want to add the optical sensor, it is very inconvenient to adjust the max spindle speed in GRBL every time you change gear ratios. I agree this is not a problem with sniffer approach.

MeJasonT, 2019-01-13:
Did I hear someone mention my sniffer - are you saying i'm a Roman (with a big nose) lol.

mmcristi1981, 2019-01-17:
I think I may be forced to go with a mega board because of the extreme memory limitations on uno. I noticed in the grbl FAQ that extra features are completely discouraged on uno.
While doing more testing I noticed something odd. It seems that bit shifting by 1 bit is very slow on 328p, like 2 to 3us for a simple x<<=1. One clock cycle is 1/16 us, any idea why it might take so many clock cycles? Maybe my measurement is not very good, but I have seen similar reports on other forums. On mega I may be able to use byte masks instead.

And the test I uploaded was incorrect, I think it worked only because I was sending numbers that didn't end with 1 on the least significant bit. It has been a while since I did bit level coding and doing that at night doesn't seem to work very well.

mmcristi1981, 2019-01-18:
Please ignore previous post about shifting, I did more testing, I was completely wrong, and tired, shifting seems very fast.

mmcristi1981, 2019-01-20:
A little update on the progress, in case anybody is still interested. So I think I'll go with SPI. Being hardware supported, it shouldn't interfere with the performance of GRBL stepping ISR. The only problem is the pins for native SPI are used on UNO. So my initial idea was to use MEGA as it has all SPI pins available. However, after some tests I think UNO's SPI can also be used: 
- MOSI=11 currently used for GRBL variable spindle PWM so can be used for SPI to send the spindle data
- MISO=12 currently used for GRBL limit Z axis, however MISO is not necessary for unidirectional SPI and can remain used for Z axis limit, this bit will be basically shifted 8 times into the SPI data register when SPI bits are sent on MOSI, I see no problem with that
- SCK=13 currently used for spindle direction, can be sent in the SPI package along with the desired speed, so direction pin can be repurposed for SPI SCK
- SS slave select can be any output pin so I can use A4 (currently not used)

I've been thinking a lot about all the solutions discussed here already, and somehow this seems the best so far. It seems like the GRBL pins were selected on purpose so that SPI might be used in case it is ever needed. In any case, if it doesn't work on UNO I am still pretty sure it would on MEGA. However I am a bit afraid to run GRBL on MEGA, as that port wasn't officially supported, not sure what is the status now. I've seen relatively recent commits there.

shooter64738, 2019-01-21:
Grbl 1.1 for mega (and the newest grbl repo) are located here:https://github.com/gnea/grbl-Mega
For some reason it just doesnt get as much traffic as the original repo.

MeJasonT, 2019-01-21:
"For some reason it just doesnt get as much traffic as the original repo"
We are stubborn old gits and Sonny cant make us move.  lol.  Not his Git repo its ours.

mmcristi1981, 2019-01-24:
Finally, a working prototype after a day of work. I uploaded the code here in case anybody is interested:
https://github.com/mmcristi1981/grbl-mega-spi
I added a readme and a screenshot of it working. I hope I didn't break any license by uploading a copy of grbl-MEGA there. I don't have time to think of another way atm.
Some important notes:
- I was forced to use the MEGA, there is no way on UNO, pin 10 is the problem. In SPI the native SS must be an output on the master board, I found this the hard way. There is also no way to easily move this limit pin to A4 for example, as it is a requirement to have all limit pins on the same port
- I haven't tried yet with longer wires. I reduced the frequency of SPI to the minimum, and the SPI test client (which you can find the source of in the repo), currently waits for two consecutive packages with the same value before it considers it a valid value. This is a very crude way of preventing errors for now.
- I kept changes clean and minimal, 99% of code is in spindle_control.c and can be disabled by commenting #define VARIABLE_SPINDLE_EXT in config.h
- Since I am forced to use the MEGA, I may end up giving up on SPI and going with one of its serial ports if I encounter any further problems down the road. But this forces me to add on the slave arduino an external USB interface since that one will have natively just one USB interface which I want to have available for updating the spindle controller code. An external USB interface would communicate to the spindle controller then using again SPI but this time over a short distance. All this only if SPI can't handle 1m of cable. Remains to be seen...

shooter64738, 2019-01-26:
Looks pretty good. I haven't used a cable 1m for spi but I have used a pretty long one (24 inches) to send lcd updates. A good shielded cable may be all you need there.

MeJasonT, 2019-02-16:
Just in case your plan falls over and as a good reference to others there is another option i had forgotten about.  Instead of the hardware serial sniffing method it is possible to have an application on your pc which splits the control of a serial port - its called a virtual serial port emulator VSPE [https://freevirtualserialports.com/](url) [http://www.eterlogic.com/Products.VSPE.html](url)
by directing the port GRBL is attached to and linking it to a virtual port you can then connect a second arduino to the virtual port and your control software.  This enables us to use the commands being sent for a multitude of different and additional controls.  Just write the parsing code to suit your needs on the second arduino.
[image]
[image]
[image]

lolsifer, 2019-02-22:
@mmcristi1981 your work has helped me out a lot.. I have modified your SPISlave to support speed control of the cheap Hyunyang inverters over RS-845... if anyone is interested I can send files - 

I am getting some erroneous speeds being sent now and again - not sure if that's a SPI transmit error?

Anyway, thanks.

mmcristi1981, 2019-02-22:
@lolsifer glad to hear that.
I assume you have a converter board from SPI to RS-485 (I assume RS-845 was a typo). This means this is also programmable somehow. Can you give more details about the setup? Would like to help you out if I can. Noise is most likely the cause for the erroneous speeds.

lolsifer, 2019-02-22:
Yeah, typo thanks..

Using this RS-845 modules to control the VFD from the UNO, running a 2.2Kw water-cooled spindle.:
http://www.yourduino.com/sunshop/index.php?l=product_detail&p=323

I have a Mega2560 -> Uno. Mega doesn't use a CNC shield as I had 4*MA860H stepper drivers left over from another project, so Mega is wired straight to the drivers..

Using bCNC on a desktop at the moment, will probably change to a pi later on..

For the moment - I'm checking it receives the same speed twice before applying.

mmcristi1981, 2019-02-22:
This is what I understood: Mega2560 -> UNO -> MAX485 -> VFD.
In that case, there are 3 comm lines where errors can occur. Do you already have a clue where out of the 3 possible places data gets lost in transmission? 

Some suggestions for each connection line:
1. Mega2560 to UNO via SPI can be corrected like you mentioned by checking for same value multiple times. Also you can try reducing the physical distance between the boards. SPI likes short distances. 
2. UNO to MAX485: this is also serial comm right? I have an idea, might be silly, but what if you remove the UNO, reducing the potential points of failure. Mega has free serials available that could be linked directly to the MAX board. Just need an extra enable pin as far as I understand the MAX board. Couldn't find yet a decent tutorial on these. This means more GRBL changes, and depends on how complex the logic in your UNO actually is. Can you post that? Would be interesting to have a look at it.
3. MAX485 to VFD: I never used RS485, but you could try to debug this link by putting another MAX485 on the other end and another arduino board at the receiving end. It also looks like the type of wire you're using is important and that you have proper termination resistors in place.

Sorry if may seem off, I am more of software guy, just trying to help. If we manage to make this work with VFDs in a reliable way, it might be interesting for others as well.

MeJasonT, 2019-02-22:
You can find information regarding RS485 here [https://www.gammon.com.au/forum/?id=11428](url)
RS485 is effectively a current loop, its connection is made using twisted pair wire such as CAT5 network cable.  for our application i would suggest screened cable as RF and EMI noise is quite an issue.  Working in industrial automation i never had any problems using RS485 however in my own projects i have experienced loads of problems, I guess the self build method lack the filtering and signal conditioning used in pro units.  As i mentioned earlier VFD's are notorious for noise, your board issues may simply be caused in the communication link at the point of its connection to the VFD.
There will be quite a bit more cod involved in getting the mega to transmit via RS485 as well as run grbl, it will be interesting to see how you approach that idea.
Consider going overboard with earthing and filtering, earth your machine and all metal surfaces independently to an earth rod hammered into the ground outside, use screened cables & attach ferrites to motor/control cables and shitty USB leads (if you don't have gold contact USB lead ie high quality with ferrites built into the cable then you really should consider it.  A mains filter/suppressor for your controller psu and house your control board inside a metal box.  This should eliminate any chance of glitches due to radio interference.  Your VFD will also have a function to change its internal operating frequency this may help to reduce the horrible squeal these things make which is also the source of the hf noise emitted from the unit. 

lolsifer - I take it your VFD has an RS485 connection for remote operation of the unit.

lolsifer, 2019-02-23:
The data is wrong between the Mega -> Uno, a low fail rate.. I've almost considered it solved by checking the data more than once.. The boards are just connected by short breadboard wires.

I do not actually have the VFD connected while I'm testing this configuration, so noise isn't causing this (at the moment?).. I'm just finishing off on this before I put it all back together again..

@MeJasonT Yeah I'm unsure of the effect on GRBL adding this in directly to the Mega.. hence separate board for now. Might be handy to have an external spindle control that also takes commands from the GRBL board.. will look into some further shielding if it becomes an issue, thank you for your suggestions..

These links should complete the picture:

_Page 50 of the manual_ 
http://www.etech.net.au/Huanyang/HY%20Series%20Inverter%2020100627.pdf
_Reference_
http://fightpc.blogspot.com/2014/10/vfd-control-with-arduino-using-rs485.html
_Reference code_
https://pastebin.com/QpmDDHsD

_Spindle/VFD on Ebay_
https://www.ebay.com.au/i/302660581322?chn=ps

mmcristi1981, 2019-02-23:
I agree that a second board adds more flexibility, where you can also override the controls of the VFD and even have your own custom control panel.
Found this regarding noise filtering for SPI, might be useful: https://www.latticesemi.com/-/media/LatticeSemi/Documents/WhitePapers/HM/ImprovingNoiseImmunityforSerialInterface.ashx?document_id=50728
If you have an oscilloscope would be great for troubleshooting this sort of problems.
I am a bit surprised because I haven't seen errors in my setup, also using just jumper wires. Maybe I didn't wait long enough (just a few minutes). Seems like you know what you're doing so me asking if you have a good ground connection directly between the two boards might seem ridiculous.

I may end up adding more code to try to make the connection more fault tolerant. I was trying keep the overhead on the GRBL board as low as possible. Still thinking about a good solution.

mmcristi1981, 2019-02-23:
A follow up, I did some more testing out of curiosity. Created a small gcode file that changes spindle speed to various values then waits. I could not see any errors in about 15 mins of testing. However as soon as I disconnected the direct ground connection between the two boards and left only the improper ground that goes through the usb ports, I got faulty values quite often.

Did the tests with both boards connected to the same PC, 4 jumper wires between them: SS, MOSI, SCK and GND. I completely left out MISO. In current implementation it's not needed.

MeJasonT, 2019-02-27:
Save the gcode file, 
Its called a spindle warm up routine, its used by professional manufacturers as a maintenance function.  If a machine has been stood for any length of time, they run the spindle up through the speed range to ensure the spindle is running ok, no bearing issues, noise, vibration or malfunction.

mmcristi1981, 2019-02-27:
Good info, thx, I think it makes sense also for accuracy, when warmed up also temperature literally increases, things change shape a little, better to have that before starting cutting. Even though for our mostly hobby purposes, this doesn't make much difference.
Just as a side note, my tests were not done yet with an actual spindle connected. Hardware is not ready yet. Just two boards connected to a notebook.

MeJasonT, 2019-02-27:
sounds like good progress.
I have my spindle speed now working over fibre optic control, i even asked my ex boss to look at an excel spreadsheet on with my speeds to get the spindle calibrated.  ended up with a third order polynomial to get a best fit. i'm 12 RPM over across the range so a little tweaking is required but no fails, no tripping the control board and no issues with rf just solid control over my spindle.
I now have a new issue, my nema 22 motor isnt man enough to hold the z axis whilst milling - I had a crash, the tool dropped.  Need to save up now and buy a nema 34 stepper motor with gearbox.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?
im using 2 stepper motors for x,y axis , solenoid for z axis , stepper motor as spindle
im really in rush for solving this problem
please

mmcristi1981, 2020-08-29:
Your details are a little sketchy, seems like a strange use case. Stepper motors are typically used in low RPM settings, where positional accuracy is relevant. Torque decreases dramatically as you go up in speed. A stepper seems like an unfortunate choice as a spindle motor.

Anyway I will try to give some suggestions:
1. use a DC motor as spindle with a DC driver if possible?
2. if (1) not possible to switch the motor type, try to connect the driver of the stepper motor for the spindle to the "z" axis output of the GRBL board. Then you control position of the spindle through g-code, rather than it's speed.
3. you need a second arduino to decode the PWM signal from the pin D11 of the Grbl board. It would then convert it to a pulse train controlled in frequency to the step pin of the driver board of your spindle motor. This is definitely not trivial and requires programming. My initial question was related to this topic, how to send the speed information from the GRBL board to a secondary arduino board in binary form. I was finally able to do that using a customized GRBL code running on arduino MEGA, using the SPI interface. But you may not need that level of accuracy and complexity. So if you can decode the D11 pwm signal from the grbl board like done here: [,](https://create.arduino.cc/projecthub/kelvineyeone/read-pwm-decode-rc-receiver-input-and-apply-fail-safe-6b90eb) but with no programming skills, this will not go far sorry...

ygjfggfgfj, 2020-08-29:
hello
for suggestion 1 it's not possible
for suggestion 2 i need to change g code instruction, and i'm not sure if
the library will understand "M3" instruction as spindle running in this case
for suggestion 3 i don't have another arduino.
do you know how to change GRBL code to handle this situation?

mmcristi1981, 2020-08-31:
hi,
sry for my late reply. Sure, it's possible to change GRBL code in several different ways. I would like to help, unfortunately I don't have the time to implement it sry, and it seems you're tight on time.

ygjfggfgfj, 2020-08-31:
Thanks for your replies and efforts.

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68

[1 more reply]