grbl/grbl #1492: ESP32 future hardware?

DE-Koodie, 2019-01-16:
I have seen the port to an ESP32 thats online, forked from grbl.
I have been using an arduino so far, but as it is at then end of its usabilty i think a new hardware is an idea. I have read that there are plans about goind to an ARM architecture in the next version, but cant find the thread right now.

Isn't going on an ESP32 the solution? 240Mhz, Dual Core. Would come handy to have one core care for I/O Serial, buttons, displays, switches, etc while the other core is up to the real time GCode interpretation? Im not a hardware professional, but my friend develops hardware and he suggested that hardware in a chat we had.

Norwegian-Gadgetman, 2019-01-16:
Or the Teensy 4.0 when it comes out?
(In Beta right now)
It does 600MHz...

The 3.6 that I'm playing with right now has 1MB Flash for program and stuff, 256KB RAM, built-in uSD-card reader, and more features than you can shake a stick at.
I expect the 4.0 to go above and beyond that...

They're 3.3V, not 5V though, but that shouldn't really matter all that much.

Slixxor, 2019-05-18:
We are using our own Spin on a GRBL port to the ESP32 (Credit to Barton Dring). Soon to be released as an OTS solution hardware wise.