grbl/grbl #1495: Spindle won’t turn on

Jamochasoup, 2019-01-18:
I have a GRBL connected to a VFD then my spindle. Recently the gcode I send to the GRBL is working except my spindle isn’t turning on. I can control the spindle with my VFD manually. There are no loose wires, everything has powe... Why isn’t my GRBL communicating to the spindle?

MeJasonT, 2019-02-16:
Please post make and model of your VFD.
I suspect you have to change settings from local to remote control.
does the VFD accept PWM signal from Grbl to drive it directly.

MeJasonT, 2019-02-26:
are you sending a spindle speed and enabling it
for example send   M3 S500  in gcode where M3 enables the spindle in CW direction and S500 is the spindle speed.
M4 = counterclockwise and M5 = spindle Off
Once a speed is set you can turn the spindle on and off using M3 / M5 until you reset GRBL of send a different speed.

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?
im using 2 stepper motors for x,y axis , solenoid for z axis , stepper motor as spindle
im really in rush for solving this problem
please

ecncshopInd, 2021-02-06:
I need this help: 

Iam using M04 command and the "spindle direction" pin out put is used for a fan on/off. 

If I use M03, then only spindle will turn on using spindle enable signal. 
If I use M04, then both spindle and fan will turn on. 

The problem Iam facing is: M05 is not turning off the "Spindle direction" signal !
It is only turning off the "spindle enable" signal. 

So Iam not able to turn off the fan with M05. 

Iam forced to give a M03 followed by an M05 to turn off the fan!

is there any way I can force both Spindle Enable & Spindle Direction to turn off , when I give M05 ?

1Metalguru, 2021-02-06:
Name Unknown:

M04 and M03 work together, and the Spindle On is a PWM output, and Spindle
direction is a direction bit which is set when M4 is used and reset when M3
is used.  M5 does not affect the Spindle Direction output as you have
found.  It remains in its previous state when M5 is used.

They cannot be used for general purpose outputs, unless you were to modify
the code and recompile.  Use the Coolant Enable A3 output to control your
fan, in conjunction with M8 g-code to turn on and M9 to turn off, I believe.

Regards,
Metalguru