grbl/grbl #1497: Grbl is not done booting

bjelinek, 2019-01-22:
Tried uninstalling and reinstalling UGS 2.0 three times and uninstalled and reinstalled grbl five times to the Arduino mega 2560. Check the Baud is at 115200 and rechecked that the port is correct. Still no joy. Then I tried SourceRabbit G-Code Sender same issue. I am not at skilled at programming, so talk to me like a 8 grader. Any help would be appreciated

109JB, 2019-01-22:
How did you load Grbl onto the Arduino?

Try using a simple serial monitor program to see if Grbl is working on the Arduino. The Arduino IDE has one built in under the "Tools" menu. You should then see something like this:

[image]

If you don't get the Grbl startup message, then either your baud rate is wrong, the com port is wrong, or Grbl didn't upload correctly.

109JB, 2019-01-22:
Also, you are on the Grbl V0.9 issues but  you say you are using a Mega 2560.  Are you using the Grbl version for the Mega? You should also be posting there since it is a Mega you are using.

https://github.com/gnea/grbl-Mega/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

bjelinek, 2019-01-23:
Thanks for the info did not realize that it is borad specific. Downloaded the mega version and everything is good now

closed 2019-01-23 by bjelinek