grbl/grbl #1498: GRBL com mensagem de erro ['SH']'SX' to unlock

tanecle, 2019-01-23:
minha cnc travou... só aparece essa mensagem:
<Alarm,MPos:0.0000,0.0
Grbl 0.9j ['$' for help]
['$H'|'SX'to unlock]... e fica repetindo seguidamente.
Preciso de ajuda.

tanecle, 2019-01-23:
[image]

MeJasonT, 2019-02-16:
Enter $H into the command line and press enter
(Digite $ H na linha de comando e pressione enter)

The default start up position is with machine locked, this is normal practice in industry with pro machines.  We are quite rightly being made to do things properly.

(A posição inicial padrão é com a máquina bloqueada, isso é uma prática normal na indústria com máquinas profissionais. Estamos, com razão, sendo obrigados a fazer as coisas corretamente.)

If translation is wrong blame google, i have no desire to make love to your car. lol.

MeJasonT, 2019-02-16:
A couple of programs you may want to look at.
GRBL Panel [https://github.com/gerritv/Grbl-Panel/releases](url)
Candle CNC [https://github.com/Denvi/Candle](url)
and Camotics [https://camotics.org/](url)
another good package is cambam not open source but well worth the price [http://www.cambam.co.uk/](url)