grbl/grbl #1501: Homing problem

MariusHotte, 2019-02-05:
Hey,
I'm using
-Arduino Uno
-cnc shield v3
-grbl v.09

I think I installed the endstops correctly. When I press the appropriate end stop in one movement, movement is stopped as well. So far everything works.
When I enter the $ H, the z-axis moves only a few inches and does not reach the endstop.

Have changed the $ parameters so far,
$ 21 = 1 (hard limits, bool)
$ 22 = 1 (homing cycle, bool)
but nothing helps. I also looked at the config.h file and changed the line like this.
#define HOMING_CYCLE_0 (1 << X_AXIS) // REQUIRED: First move Z to clear workspace.
// #define HOMING_CYCLE_1 ((1 << X_AXIS) | (1 << Y_AXIS)) // OPTIONAL: Then move X, Y at the same time.
// #define HOMING_CYCLE_2
 From now on, only the X-axis should search for its home position.
Still have the same problem that the axle moves only a few inches and does not reach its end stop.
I know which parameters I should or can change.
Hope you could help me. Thanks in advance

cfoxwood60, 2019-02-18:
Does it give you an error code when it stops. I had a similar problem when I tried to run a program. I solved it by making all the limit switches normally closed. Hope this helps

cfoxwood60, 2019-02-18:
Im sorry I am using 1.1 and I dont think you can use normally closed limit switches on your version.

MeJasonT, 2019-02-26:
If $20 is enabled ie $20=1 and you exceed the machines workspace it will also trigger an end stop

$130, $131 and $132 are the size of your work area X,Y,Z
$24 and $25 should have a feedrate entered by default is $24=25  $25=500

If you have these settings correct then the axis should seek home
as you have hardware end stops you could try disabling Software limits $20=0 and see if it allows you to seek home.

Be prepared to kill power to your motors just in case anything goes wrong whilst setting up your end stops or use the reset button of your GUI.

just for reference, end stops can be inverted Re cfoxwood60's comment

[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#5----limit-pins-invert-boolean](url)
`$5 - Limit pins invert, boolean
By default, the limit pins are held normally-high with the Arduino's internal pull-up resistor. When a limit pin is low, Grbl interprets this as triggered. For the opposite behavior, just invert the limit pins by typing $5=1. Disable with $5=0. You may need a power cycle to load the change.

NOTE: For more advanced usage, the internal pull-up resistor on the limit pins may be disabled in config.h.`