grbl/grbl #1502: Auto Square Setup

zurpv1, 2019-02-06:
Hi,
I am using a CNC Shield v3 with GRBL 1.1. I have dual Y motors, one being cloned from driver A. I have an end stop on the Y motor that is using the Y driver. I would like to add another end stop switch to the second Y motor that is cloned via driver A. How do I wire that and are there any setting that I need to change in GRBL?

I apologize for posting this as an "issue", however, I didn't see any other place to post a question.

Thank you for your help.

jahnj0584, 2019-02-06:
You can't at the moment, but it'd be nice if you could.

MeJasonT, 2019-02-26:
You have 2  Y axis motors ?
Is your second motor performing another axis ie A or B axis

There isn't any need to touch GRBL, it sounds like you need to connect both Y axis end stops in parallel if using Normally open contacts (else serial connection for NC contacts).
This will stop the machine if either end stop is triggered.

jahnj0584, 2019-02-26:
Lots of machines have 2 Y screws. A and B axis are rotaries.

You don't understand how squaring works. You need to move a single motor
after the other one already triggered TJ switch.

Norwegian-Gadgetman, 2019-02-26:
NO!
You do NOT want 2 y-axis endstops(I'm assuming that you don't mean one at each end of one of the y-axis, but one at 0 at both y-axis).

If you have the two y-axis motors connected correctly the motos will ALWAYS stay in step, so the same endstop should always activate. If that doesn't happen it muct mean that your motors are allowed to get out of step and that there's enough flex in the machine to allow it to happen.    
This is bad...
First of all, it means that REPEATABILITY is a word you can toss out the window. 

Your Y-axis should be connected as seen on this page;
http://shapeoko.github.io/Docs/wiring.html
If you remove the driver, and only keep the motors hooked up to each other you should find that if you rotate one, ten the other will also rotate the same way.   
As long as they're connected together, one will not get out of step with the other unless you really, really mess up. (It takes force to make them get out of step)

If the machine isn't completely square, then you just have to make it square. 

On my ShapeOKO (first gen F-series, only came with one y-axis motor) this took a while...

I built a torsion box table and mounted the y-axis rails on top. 
I adjusted one of them to be as level with the surface as possible, That's my 'base'. 
The other Y-axis was adjusted to be parallell to it(measuring both directly across, and diagonals)
Then I used a GOOD spirit level (less than 0.5mm pr 1meter error. This should be listed on it. If it isn't, its junk) to verify that it was exactly level with the first. 
Then I installed the carriages and the X-axis. And adjusted the x-sxiss to be exactly 90degrees to the first y-axis. To get it right I used very, very thin shims between the carrier plate and the end of the rails. 0.01mm thin. (They sell stuff like that on ebay. In fact you can find even thinner)
With the X and Y sorted, I tightened the belts, and used a guitar-tuning app on my phone and twanging the belts to verify that they had the same tension.   
Now I actually ran the machine for a little while, with a endmill, and carved an L-shaped area in the wasteboard. This L-shape would be exacly level wirth the machine, so now I could place a square onto the branches of the L and check the alignment of the Z-axis, the spindle and the motoer plate on the X-axis. 
After that I skimmed the wasteboard with a lare endmill. 

It took time, but its what must be done to get a machine square and level. 

A lot of the process, and pictures can be seen in my build log...
https://forum.shapeoko.com/viewtopic.php?f=5&t=2025
Lots of nice pictures on page 15 and onwards, including what I intend to do with he X-axis carriage and the Z-axis and the spindle mount.  
I'm currently at 0.15mm accuracy, and will be pushing it towards 0.1mm or better.

MeJasonT, 2019-02-26:
jahnj0584 show some respect, the question was not clear.
The second part of my answer is in agreement with the statement from Norwegian-Gadg
>NO!
You do NOT want 2 y-axis endstops(I'm assuming that you don't mean one at each end of one of the y-axis, but one at 0 at both y-axis).

J-Dunn, 2019-05-11:
I have a twin motor axis and considered modding grbl to home both drives together to a primary home switch, then tweak in the second one ( jogging just one motor ) to detect a second switch which is mechanically set up to ensure axes are perpendicular. 

I started to question how repeatable the switches would be for this to be reliable and consistent at each start-up cycle.  I concluded it would be mechanically more stable ( and simpler ) to link the drives with a toothed belt and set the machine up once then lock pulleys in place. 

To do this I bought dual axis versions of my stepper motors ( 20mm of axis out of rear of motor ). This allows fitting a thoothed pulley to each motor. By pure luck the distance between my motor axes plus 20T pulleys worked out to need a 1600mm belt: a standard size, so I did not even need a belt tensioner !

This puts a small side load on the motor bearings but since there is not a lot of tension on the belt, I don't think this will cause a problem for the bearings. ( stepper motor bearings are quite small and should not be loaded.)