grbl/grbl #1503: Having trouble with setting physical origin (0,0,0)

pilant, 2019-02-06:
I've been having confusing time trying to get my (from Amazon) ubiquitous "CNC 3018" machine going.  I can get it to "work" (jog) from the grblControl console, but when I actually try to have it use the "iPhone" example, it just doesn't work (see below).

What I ultimately reduce the issue to is this: when I issue (to grblControl) the G-code command "G90 G0 X0", the spindle assembly is positioned to about 60% from the left edge of the platen (mesa).  "G90 G0 Y0" positions to about 60% from the bottom edge, and "G90 G0 Z0" tries to position above the spindle assembly.  So it would appear the physical machine origin is incorrect.

To make sure I had the direction correct, I used the "G91 G0 X1" and set the $3 direction mask to move the stepper in the proper direction to cause the spindle assembly to move to the right.  I did the same operation for the Y and Z axis.  Resulting in $3=7.

Having installed limit switches, I can get homing to operate, but the resulting machine coordinates are negative.  (Possibly due to the machine coordinates being incorrect.)  Which seems to cause all manner of problems.  (In particular, the "iPhone" example causes an alarm when the limit switches are activated.)

So I'm wondering if anyone can spot / point out my stupid mistake?

Thanks in advance.

- Mark

jahnj0584, 2019-02-06:
You still have to jog and reset work zero after homing

pilant, 2019-02-06:
> You still have to jog and reset work zero after homing

After homing I can jog as much as I want (away from the limit switches) and the spindle assembly moves as expected.  BTW, I can "disable" the limit switches ($21=0 and $22=0) without any change in behavior.

This brings up another possible clue: after homing the work coordinates are all zero as expected.  However, when I jog in any direction (away from the limit switches) the appropriate stepper will operate, but the work coordinates never change from all zeros.

- Mark

pilant, 2019-02-06:
Forgot to add, the board is running grbl 1.1f

pilant, 2019-02-07:
A bit of an update.  I figured out why a "G90 G0 X0" was only going part way across the table.  It would appear, contrary to the Wiki (documentation) indicates, the max travel settings do affect the location of the machine zero.  When I changed them to 300.0 and 130.0 (for X and Y), using G90 G0 X0 Y0 went to the edges of the table.  (Although it was the upper right corner of the table.)

Here are my current settings:
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=7
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=800.000
$111=800.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=300.000
$131=180.000
$132=40.000

With these settings, the jog buttons on the grblControl panel operate as expected, moving the steppers to allow the spindle assembly to move in the correct direction.  Also, clicking the home button moves the spindle to the lower left corner of the table, and the spindle assembly lowered.  Using the G91 incremental distance command, to move in the X, Y, and Z directions, all work as expected.  (Assuming a lower left table corner as X=0, Y=0, and spindle down as Z=0.)

What is very curious is the machine zero is the upper right corner of the table, the work coordinates never change from zero, and the Zero XY, Zero Z, Restore XYZ, and Safe Z are disabled / greyed out.

All the videos make it look so easy... but having spent the past several days trying to get this sorted out is a bit frustrating.  I'm wondering what I'm missing... probably something very simple.

- Mark

jahnj0584, 2019-02-08:
Try UGS instead. I've never used Panel and don't remember it being
compatable with the new grbl.

pilant, 2019-02-08:
Well, I finally got it working.  Short answer: it appears grblControl_0.8.1 (running the pre-built version) doesn't work very well on my host system (Windows 7 Ultimate).  The solution was to download (the pre-built version) of Candle 1.1.7 and all is working.  BTW, this is with the original settings without any of the limit settings changed.  (That is next, because I like the idea of having limit switches :-)

Longer answer: I had to do a bit of a learning in terms of CNC terminology.  In particular absolute movement being based on the zero point set by the CNC operator... NOT (0,0,0) based on the physical location on the table.  Once I understood this I realized (most likely) my problem with grblControl was not being able to reset the local XYZ because the buttons were disabled.  This was compounded by not having the work coordinates updated.

At the suggestion from @jahn0584 (thank you very much) I decided to try some other programs.  GrblPanel was the first I tried, and it worked fine.  (Although I did miss work rendering frame.)  Quite by accident, I stumbled on Candle and decided to give it a try.  It worked very well, and had features I couldn't get working in grblControl working fine.

So now to finish getting everything together and actually give it a try doing a real project.

- Mark

closed 2019-02-08 by pilant

georgevitale, 2020-05-12:
Hi Mark ! It seams I'm having the exact same problem and can't figure it out. My Machine coordinates are showing 0,0,0 and wont change when I Jog. 
Can you give me some detail as to what you did.
Thanks George.

pilant, 2020-05-16:
Hi George.  Sorry it took so long to get back.  It has been a while since I looked at this, so my memory is a little dim.  As I recall, I never had any problems with the coordinates changing when the actual position changed.  One way I found to get a better handle on what was going on was to look at the actual serial commands being sent to the CNC (in the command / console area).

As I mentioned,I believe my biggest problem was with running grblControl.  I attempted to verify this by using another grbl program (Candele) to duplicate what I was seeing.  Since I wasn't able to duplicate the "problem" I just assumed it was "something" in grblControl.  As I didn't feel like climbing into the code to determine the root cause of my problems, I just switched to using Candle.

I hope this hlps.

\- Mark