grbl/grbl #1505: Issue with limit-switches

Averell64, 2019-02-16:
Hey,
I've got an issue where when I connect all the limit switches to the cnc shield I'm not able to control to the board anymore even though the switches are not triggered (I'm using N/O switches)... I can connect them while i have the board running and connected to the pc without any problems but as soon as I trigger one of the switches everything stops and I have to powercycle the board (which is how it's supposed to be) but after that I'm not able to properly connect to the board anymore.... I'm using GRBL controller 1.1 and there it would only say "Grbl 1.1g ["$" for help]" and not send the two starting commands "$G" and "$$" anymore and I cant send any commands on my own.... I have $21 enabled and it worked fine before with only one endstop which i used as an emergency switch during testing and dialing in the steps per mm on the motors but as soon as i started using all 3 switches and disabled the spindle speed thingie in the grbl library to use the z endstops on the cnc shield it stopped working properly. I can still use the machine without endstops but its annoying to have to pay so much attention to the machine not crashing and always setting everything up manually... Grbl controller also wont show me the work coordinates and or machine coordinates even though I put the setting to always show...
It would be awesome if anybody knows a solution to this issue.

~Averell

ps: I'm quite a newbie when it comes to electronics n stuff so it'd be nice if you explained everything for idiots ;)

MeJasonT, 2019-02-26:
load the code back onto your grbl again, i suspect that the processor has lost its program or has ended up in an unrecoverable state.
spindle speed thingie ? are you using a controller which was configured for GRBL ver 0.9 so your pins are not configured correctly - that's the only reason i can think of as to why you decided to make changes in the code.

In the $$ settings did you invert any pins, from what you describe it sound like the limit switches are NO but the code has inverted them perhaps.

Get back to basics with the board working again by re flashing it and we can talk you through the setup from there.

Averell64, 2019-03-03:
I actually found the issue - I had some interference from the motor cables as I had both, motor and endstopcables in the same cabletube... I isolated the motorcables and now everything works like a charme. Thanks for the answer though MeJasonT (with "spindle speed thingie" i meant the z+ and z- pins on the shield which by default are not z+ and z- but rather the pins for spindle control, and since i suck at soldering and use a dremel anyway i just deactivated the spindlecontrol so i could use the pins for the endstops) :)

closed 2019-03-03 by Averell64