grbl/grbl #1506: Pulsing Y Axis

FutureLights, 2019-02-22:
I am having an issue with my Y axis moving in "pulses". It will go stop go stop the entire time when I try to move it with something like G1 Y10. There is no pressure on it and my X axis it working smoothly. I have triple checked that all my wires are connected. Could I not be giving it enough power?  I have tired changing the feed rate with no success. I stopped working on this project a few months ago and it was working then. This is not the only issue I have but it seems to be the biggest. I am new so I'm sorry if this is a simple question or in the wrong spot.

chamnit, 2019-02-22:
Depending on what stepper driver setup you have, they will often have a small potentiometer on the driver that sets the output current. Yours may be low.

FutureLights, 2019-02-22:
I'm using Easy Drivers for my stepper motor drivers. Interestingly when I increase the potentiometer the stops and starts are very pronounced, but when I decrease it, it blends together and almost like wavers getting stronger and weaker. Should I record the audio?

terjeio, 2019-02-23:
Could it be the PSU overload protection kicking in? Some (or most?) switch mode PSUs will deliver power in short pulses when overloaded.

langwadt, 2019-02-23:
or the drivers overheating and shutting down

FutureLights, 2019-02-23:
My X Axis Driver has no issues like this. The red LED still stays on while it runs.

jahnj0584, 2019-02-23:
It's none of the above things. Take the motors off the machine and see if it persists. Then you know if it's a load setting or not. Try with 1/2 the acceleration value.

FutureLights, 2019-02-23:
Thanks so much for the help, I would have no idea what to do. I don't know what the PSU overload protection is but the driver does not seem to over heat or turn on and off and the indicator light is on always on while it's move (no blinking). I don't know if just part of the driver could just turn off and on though. I tried disconnecting the motor from the machine and the issue continued, even when I removed half of the acceleration. Could I have changed some setting in Grbl to make it do this accidentally?

FutureLights, 2019-03-05:
I've done some more testing. I removed the wires giving power to the x axis to see if it was using up too much power. The issue still continued. I tried 4 different power supplies and they all worked well in powering the x axis. I switched the wires so I'd use the x axis stepper driver to power the y axis motor and it worked. Could I have done something to my y axis driver that would make it behave this way?