grbl/grbl #1507: GRBL stalls with Bf:14,0

Elmi77, 2019-02-22:
Working with the most recent version from GIT I stumble upon a strange problem. I'm quite sure I'm doing something wrong but have no clue what it could be. I stream my G-code to GRBL and check regularly the buffer fill level by sending a "?" and evaluating the result. My code ensures the command queue never becomes full. But at some point I get back a

Bf:14,0

which means one command is in queue and the serial buffer is filled up completely. Amazingly at this point the serial buffer is not processed any more, no matter what I do I do not get rid of this state and have to restart my hardware.

So...any idea what could cause the serial buffer to lock when it is filled completely?

Thanks!

MeJasonT, 2019-02-22:
From [https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface](url)
for some reason your controller is not expecting to receive anything.
I'm not sure why this would be, have you changes any parameters which has altered the buffer values/message size.

let us know what front end software you are using.

> Buffer State:

Bf:15,128. The first value is the number of available blocks in the planner buffer and the second is number of available bytes in the serial RX buffer.

The usage of this data is generally for debugging an interface, but is known to be used to control some GUI-specific tasks. While this is disabled by default, GUIs should expect this data field to appear, but they may ignore it, if desired.

NOTE: The buffer state values changed from showing "in-use" blocks or bytes to "available". This change does not require the GUI knowing how many block/bytes Grbl has been compiled with.

This data field appears:

In every status report when enabled. It is disabled in the settings mask by default.
This data field will not appear if:

It is disabled by the $ status report mask setting or disabled in the config.h file.<

109JB, 2019-02-23:
Since you state:

>But at some point I get back a

>Bf:14,0

>which means one command is in queue and the serial buffer is filled up completely.

I suspect you are using V1.1 of Grbl since the buffer state messages changed and the above description would match the V1.1 reporting.

I have experienced a similar problem, but not on the "official" V1.1 release for the Arduino.  I have been working with the V1.1 running on the ESP32 and experienced the problem on that platform.  In my case it happened during jogging and occasionally the following would occur:

1. Jog issued. Command was to end of machine travel at max rate 5080 mm/min
2. Cancel jog realtime command sent
3. The machine speed slowed significantly, but did not stop. Slows to something like 75 mm/min.

At this time the status reports show Bf:14, 128 because I was using a send-response type protocol.  If i try to issue another jog cancel which is a single character, the report then shows Bf:14, 127. So the character gets to the serial buffer and Grbl knows it is there but something is preventing Grbl from moving past the command that is taking that one planner buffer block.  If I continue to send commands, it keeps driving the serial buffer down consistent with the length of the commands sent. There is no 'ok' response or otherwise from Grbl except the serial buffer updates in the status reports.  I could easily drive it down to BF:14,0 as described in the original post.  

I tried to duplicate this on the Arduino version but haven't had it do it on the AVR Grbl yet.  I can get it to happen occasionally on the ESP32 version.  Here is my bug report over there.  

https://github.com/bdring/Grbl_Esp32/issues/91

Something is going on that isn't easily repeatable and it sounds like it may affect the AVR version as well as the ESP32 version.  I am going to try to investigate more to see if i can pinpoint what sequence will cause it to act up.  Unfortunately I don't have the C/C++ skills to examine the code itself, but maybe I can figure out what command sequence is causing my problem which could help someone pinpoint what may be going on with the Grbl code.

MeJasonT, 2019-02-24:
does your controller have a CH340 usb chip? is your board a clone ?
[https://www.cnczone.com/forums/arduino/349176-cnc.html](url)

hey John you cant be expected to remember everything.

Elmi77, 2019-02-25:
To answer the most important question: yes, I'm using v1.1, seem's I accidentally posted this to the wrong project - I'll reopen this issue in correct repository.

Just to be complete: I do not use neither a standard Arduino nor a GUI, I'm communicating with the MCU via serial interface directly out of an own application - that's why I suspect I'm doing something wrong here.

MeJasonT, 2019-02-25:
This is going to get confusing, do we answer here where all the postings are or randomly add a comment to your new post, i decided the sandwich filling was here.
Ok so your serial port in your application is set for 115200,n,8,1 (no parity, 8bit, 1 stop bit) ?
using a carriage return at the end of your code transmission, also works with linefeed and carriage return but responds with ok ok instead of just one.  How often are you polling GRBL for a status ie sending "?"
every 200ms is acceptable, Candle by default polled every 50ms and caused me heaps of problems.
It sounds like its the speed in which you send commands that's causing you issues.  GRBL will quite happily transmit status reports in its own time when requested if you haven't changed the firmware.

chamnit, 2019-02-25:
Please provide your Grbl $$ settings, $I build info output, and the complete status report at the error.

Elmi77, 2019-02-26:
> your serial port in your application is set for 115200,n,8,1 (no parity, 8bit, 1 stop bit)

It is running with 57600 bps

> polled every 50ms and caused me heaps of problems

This could be the problem - after adding a delay before every transmission the deadlock did not occur any more. But to be sure to not to get locked just because data have been sent too fast accidentally - is there a fix available for this problem?

chamnit, 2019-02-26:
It takes both time and resources to generate status reports. I don’t recommend querying more that a handful of times per second at most. While it looks like Grbl can do faster, you begin to take resourced from the rest of the main program and will effect how the system runs.

Elmi77, 2019-02-27:
I fully understand this and I also would accept that GRBL does not react/react in an unpredictable way as long as requests are made too fast. But my problem is the deadlock, once I'm in this "Bf: 14,0" state, it is impossible to communicate with GRBL, any more also when there are no more requests sent in a way too high frequency. In this state a reset has to be issued in order to get back control. That's something where I'm looking for a solution now, just for the case when GRBL gets into this state accidentally.

MeJasonT, 2019-02-27:
It will not happen if you get your communication in sync, if people on TV are talking really fast and you miss what they are saying then you have a fair idea what is happening to your setup.  What you are describing is a fault caused by your GUI, if you wish to have the issue not happen then you require a means of controlling your data flow.  The unpredictable state you mention is a fault condition, having a soft reset makes no sense - the GUI/sender has caused a data overflow and hung the processor.  The best solution is to write a better communication handler for your GUI, if GRBL is communicated with as designed it works perfectly well in 99.9% of cases.

Elmi77, 2019-02-27:
> It will not happen if you get your communication in sync, if people on TV are talking really fast and
> you miss what they are saying then you have a fair idea what is happening to your setup.

I understand this and I agree with you. But my point is something different: when people in TV start talking slower at some point, I have a chance to understand what they are saying then.
But the current GRBL-behaviour is different: no matter how slow the people in TV talk, beyond this "point of no return" my TV is unusable until I turn it off an on.
So It is no problem for me that during a way to fast communication there is no (usable) response, but I think it is problematic the communication is locked forever, also when the GUI slows down and tries to communicate in a normal way.

MeJasonT, 2019-02-27:
I want to come across as non argumentative and wish to express why you cant have anything other than a  total reset, the micro has stalled, computer says no. You wouldn't really want a button you could magically press to get back control as you would get nothing but gibberish and erratic control back.  The code will probably at some point where its in the middle of doing something and you really wouldn't. want to return to that same point.  Its always better after a crash or major fault to stop, reset and start again.  If your were half way through milling an arc and started again you would get 3/4 instead of a semi circle for example.  Or you could get high pitched screaming, big mechanical bang sounds and a distinct ping as you duck to avoid the tool escaping from the collet.  Reset is your friend.

Elmi77, 2019-02-27:
@MeJasonT when this communication problem happens, there is no problem with the movement! The commands which have arrived at the controller prior to the lock are processed correctly, the only problem is, I can not send any more commands as soon as the communication interface has locked. So without this lock it would be possible to continue properly just by sending the next commands (slowly).

Anyway, thank you for your help!

MeJasonT, 2019-03-03:
Does this relate to your issue
[https://github.com/grbl/grbl/issues/1504](url)