grbl/grbl #1510: x axis only spins one way

Slawkos, 2019-03-05:
Hi,
I am building CNC router
The hardware:
Clone UNO
CNC Shield v3.0 with 4 x A4988
4 x TB6600
4 x 23HD76002Y-21B1 Busheng bipolar
PSU 12V 20A
The software, tried all of them
grbl_v0_9j_atmega328p_16mhz_115200.hex
grbl_v1.1e.20161208.hex
grbl_v1.1f.20170801.hex

Control:
ugsplatform-2.0.0-stable2
wired as on pics

[image]

x goes only one way, on the TB6600
when on A4988 goes booth directions
acts the same on all the drives and motors so none of them are burned
checked the voltage on pins:
x step 29mV idle, to 100mV when x+ or x- jog
y step 31mV idle, to 400mV when y+ or y- jog
z step 31mV idle, to 190mV when z+ or z- jog
x drive 1.26V when jog x- and 4.5V when jog x-
y drive 4.5V when jog y+ and 30mV when jog y-
z drive 34mV when jog z+ and  5V when jog z-

any suggestions please
regards
Slawkos

MeJasonT, 2019-03-05:
check the order of your motor wiring  A+ A- B+ B-

Slawkos, 2019-03-06:
I have all possible variations done already, unsuccessful

MeJasonT, 2019-03-06:
I have a couple of ideas you could try.
look on tinterweb for your motors datasheet and see what wire colouring they have used for the pairs
or with the motor disconnected try to find the pairs by selecting one wire and trying it to each of the others and rotate the shaft by hand, you should feel a resistance against you rotating it when you have a pair, the remaining two wires will be the other pair.  you have 2 coils A and B hence A+ A- B+ B-
if for example you have wired A+ A- B- B+ (second pair reversed) the motor will only run in one direction what is happening is you get one step then an effective electronic brake then the next step moves it again a half step.  when the stepper command is reversed with the second winding reversed the step is made in the original direction and the second half step is again blocked.
That is what you are experiencing.  The motor works , you have proved that using the other stepper driver.  I cant say the problem you are having is easy to fix but it is definitely the the motor wiring.
It happens to us all at some point or other, you will kick yourself when you discover what you have done.
don't assume the step stick and the stepper driver wire in the motor in same order.

Slawkos, 2019-03-06:
Thank you for your time and help, I have done that already as I was aware of it before, have also checked the resistance on the cables and it is 1.2ohm on each coil, also have got schematic of the motors. I think it might be the voltage 1.26V on the x drive pin as it's higher than on y and z. It is very strange
[image]

109JB, 2019-03-06:
> if for example you have wired A+ A- B- B+ (second pair reversed) the motor will only run in one direction 

Not true.  The motor in question is a 4 wire motor and as long as the A coil wires go to the A terminals on the driver in either order, and same for the B wires, the motor will run just fine. As a matter of fact, in older controllers, this is actually what you would do for inverting the direction because some of that older software didn't have anything like a $3 invert dir pin setting. I actually do this on all of my machines where this is possible once I get them up and running so that $3=0 is correct. Only closed loop stepper drives really care if B+ and B- or A+ and A- oriented a certain way because the motor and encoder directions have to match.

@Slawkos - What are your Grbl settings?  

I would set $1=255 for sure to test. 

There is also a config.h setting for step pulse delay. Some drivers require a delay between setting the direction pin and initiating the step pulses. On those drives that need this, if no delay is set the drive may not have time to set the direction.

Slawkos, 2019-03-07:
Hi, thank you for your help
my settings:
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 1600.000    (X-axis travel resolution, step/mm)
$101 = 1600.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 200.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)

tried to change in config.h but compiling with errors so cant upload, changed in firmware settings instead.
after changing $1=255 nothing has changed, tried many other values with no luck

jahnj0584, 2019-03-07:
One direction usually means a weak DIR pin connection. Also, what is the
device and is 500mm really the top speed?

109JB, 2019-03-07:
Try it connected directly to the Arduino without the shield to see if it is a problem with the shield

Slawkos, 2019-04-03:
A big thank you :) for your help folks, the problem was faulty Arduino (it took some time to arrive to me from china) and after programming new one all works fine :)

closed 2019-04-03 by Slawkos