grbl/grbl #1514: Drive controller with step left or right only

Elvis999IT, 2019-03-18:
Hi
I have an old motor controller with Step-right and Step-Left, so no direction and step...
There is a way to use it?
Is possible made some change in the software to use step-left and step-right instead of step and direction?

cri-s, 2019-03-18:
Yes, depending on direction, either the step or direction pin is used as
pulse. It should be clear that step-mask should include the direction pin
too in order it works.

cri-s, 2019-03-18:
Now i'm home, here the extract from the code:for y axis, the other are equal.

inside cpu_map ...
	#define STEP_Y	CW_CCW
	
	#ifndef STEP_Y	
	 #define STEP_Y  STEP_DIR
	 #define Y_CCW_BIT 0
	#else
	 #define Y_CCW_BIT Y_DIRECTION_BIT
	#endif

  #define CCW_MASK
((1<<X_CCW_BIT)|(1<<Y_CCW_BIT)|(1<<Z_CCW_BIT)|bit(Y_CCW_BIT)|bit(A_CCW_BIT))
// All CCW bits
  #define STEP_MASK
((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)|(1<<A_STEP_BIT)|CCW_MASK)
// All step bits
  #define DIRECTION_MASK
(((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)|(1<<Z_DIRECTION_BIT)|(1<<A_DIRECTION_BIT))^CCW_MASK)

inside stepper.c
	#define STEP(x)	CONCAT(STEP_,x)(x)
	#define STEP_DIR(x)	bit(CONCAT(x,_STEP_BIT)
	#define CW_CCW(x)	st.exec_block->direction_bits &
bit(CONCAT(x,_DIRECTION_BIT)) ? bit(CONCAT(x,_DIRECTION_BIT)) :
STEP_DIR(x)
	

code snipped for step/dir or cw/ccw, just replace (1<<Y_STEP_BIT) with STEP(Y)

  #ifdef ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING
    st.counter_y += st.steps[Y_AXIS];
  #else
    st.counter_y += st.exec_block->steps[Y_AXIS];
  #endif
  if (st.counter_y > st.exec_block->step_event_count) {
    st.step_outbits |= STEP(Y) ;
    st.counter_y -= st.exec_block->step_event_count;
    if (st.exec_block->steps_backlash[Y_AXIS])
st.exec_block->steps_backlash[Y_AXIS]--; else
    if (st.exec_block->direction_bits & (1<<Y_DIRECTION_BIT)) {
sys_position[Y_AXIS]--; }
    else { sys_position[Y_AXIS]++; }
  }

2019-03-18 19:11 GMT+01:00, Cri S <phone.cri@gmail.com>:

milesmiles902, 2023-04-28:
Repeatedly, STEP/DIRECTION vs CW/CCW wiring schematics seems be a function of frequency, rather than wiring.

I'd re-evaluate the master branch of GRBL to better fit your frequency and pulse train.