grbl/grbl #1515: alphabet W Z X V N M K W Y benting or distort. no straight line.

winson2769, 2019-03-23:
hi all pls help, i try to used single line font to engraving my name,but i can get the straight line for alphabet W Z X V N M K W Y .they are all benting or distort in the middle half way.i am using cnc 3018 machine candle 1.17 version. i already run the machine at very slow speed but still same.is that my machine setup problem or gcode programming problem?is that anyway to make it straight? thanks

PicEngraver, 2019-04-03:
It appears to me that you probably have a mechanical problem with your machine.  Something is maybe loose or bent and allowing the bit to wander a little.  It is not a grbl problem.  Since the same letters do not repeat the same pattern, it is not the gcode file or the font you used.

winson2769, 2019-04-04:
thanks for your adviced.

PicEngraver, 2019-04-04:
Please let us all know what you find.
Good luck.

winson2769, 2019-04-05:
hi i guess i removed away the end corner screw holding bracket for the x-axis motor  , readjust the horrizontal machine plate, change font and become normal and better.