grbl/grbl #1517: GRBL on arduino nano

roydekleijn, 2019-03-25:
Hi,

I like to install GRBL on a Arduino Nano.. In my project I only need 2 steppers + 1 servo (for Z-axis movement) + 2 limit switches

I get the following error: 

```
Sketch uses 30880 bytes (100%) of program storage space. Maximum is 30720 bytes.

Global variables use 1472 bytes (71%) of dynamic memory, leaving 576 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; 
```

Is there a way I can comment some code? since I only need 2 steppers and 1 servo + 2 limit switches.

I hope this is possible, since changing the bootloader gave me different issues.

Thanks

Norwegian-Gadgetman, 2019-03-25:
Is there a specific reason you're using the nano?

If it's the size that's the issue, you could go for a Teensy instead. The Teensy 2.0 is about the same, but the Teensy++ 2.0 has 4x the space but is slightly longer.
(The 3.x series are kind of silly overpowered with up to 1MB program space and clock speeds in the 120Mhz+ ranges, and also mostly 3.3V devices, so just ignore those... )

<https://www.pjrc.com/teensy/>

MeJasonT, 2019-03-25:
It loads to Nano's ?  Are you using GRBL IDE to load the code or XLoader.
You could try Xloader if you haven't already its a cleaner more direct way of uploading the code
[http://www.hobbytronics.co.uk/arduino-xloader](url)
Have you used the correct settings in the IDE before uploading - It sound strange why it wont work for you.

MeJasonT, 2019-03-25:
> Sketch uses 30880 bytes (100%) of program storage space. Maximum is 30720 bytes.

What version are you using, this seems to be a huge code for the ATMEGA328 IC based GRBL

roydekleijn, 2019-03-25:
Thanks for the comments!

I have the 328P version. The reason I use a nano is that I'm building a mini CoreXY plotter with this board: cnc shield V4 (I don't think Teensy fits on this board)

I uncommented the following 2 lines:

```
#define COREXY // Default disabled. Uncomment to enable.
#define VARIABLE_SPINDLE // Default enabled. Comment to disable.
```

and I changed the homing settings to:
```
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```

1. because I use CoreXY
2. because I like to use a servo for lifting the pencil

If I comment `#define VARIABLE_SPINDLE // Default enabled. Comment to disable.`, then the CoreXY is working fine (including homing). But I'm not able to lift the pencil.

MeJasonT, 2019-03-25:
Are you changing #define VARIABLE_SPINDLE because your hardware is configured for ver 0.9 of grbl
[https://github.com/grbl/grbl/wiki/Connecting-Grbl](url)

roydekleijn, 2019-03-25:
I was changing `#define VARIABLE_SPINDLE`, because I assumed that it would allow me to control the servo with the `M03` and `M05` commands.

Are my thoughts wrong here?

MeJasonT, 2019-03-25:
The spindle output by default is pwm for variable spindle, i believe there is some trickery involved with setting up for lasers but out of the box it should do exactly what you want M3 and M5 will control a spindle with speed set by S*** speed (gcode - M3 S1000 will start a spindle running at 1000 Rpm all be it that the output signal will be 0-255 pulse width.) 

I'm confused by the requirement of a spindle when you are using a plotter pencil.
converting the z axis from step to servo control for up/down of the pencil is more the issue you have.  Actually you are building a 3 axis machine but the Z isn't required to have variable height?
There will be examples on the web where people have built plotters and changed the code to operate a servo or a solenoid. definitely worth a look.
[https://github.com/svenhb/GRBL-Plotter](url)
[https://www.instructables.com/id/How-to-Control-a-Servo-Using-GRBL/](url)
[https://www.google.com/search?q=using+grbl+as+a+plotter&oq=using+grbl+as+a+plotter&aqs=chrome..69i57.12449j0j4&sourceid=chrome&ie=UTF-8](url)

roydekleijn, 2019-03-25:
@MeJasonT Thanks, maybe I misunderstood the implementation... I will read  through your links and see if that helps for a solution.

roydekleijn, 2019-03-25:
Weird, my servo is not moving on the M3 and M5 commands..

MeJasonT, 2019-03-25:
Please can you send details of the servo you are using and how you are connecting it.

roydekleijn, 2019-03-26:
Sure, I tried both SG90 and MG90S. Connecting to 5V and GND and the yellow wire to pin Z+ and Z- (both gave the same result) of the cnc shield v4.

Do I need to change something in config.h? 

The cnc shield v4 has also a A6,A7,D12,D13 maybe that can be of some use?

My last try was a default GRBL with only `#define COREXY ` enabled and correct homing setting.

MeJasonT, 2019-03-26:
umm interesting, is this your board by any chance.
[https://www.youtube.com/watch?v=1UKY1_hOipQ](url)

roydekleijn, 2019-03-26:
yes, that is my board!

MeJasonT, 2019-03-26:
Has the guy in the second link explained how he got his servo working.

roydekleijn, 2019-03-27:
I like to point also to this thread: https://github.com/bdring/Grbl_Pen_Servo/issues/1#issuecomment-477002412

MeJasonT, 2019-03-27:
Have you tried any of the suggestions made in that thread.  It sounds like the board programmed to run just a servo code works using -Z pin (pin 11 which also corresponds to GRBL Spindle ver 1.1).
It sounds like the guys in link [bdring/Grbl_Pen_Servo#1 (comment)](url) are using version 0.9 of GRBL as they are talking about the Z End stop.  Just make sure you are not falling down their hole.  I don't know what effect pwm has on the servo the two signals are different but many have had success without changing anything.  If you happen to have a uno lying around or you can breadbord the nano so that you can connect the servo without the CNC board, Upload GRBL 1.1 to it and see what happens when you send an NC file containing Z moves.  Keep the default settings - grbl will quite happily run without any hardware connected, as soon as you change the settings you end up needing limit switches connected and to home imaginary axis.  Keep it simple, you can then alter Z axis acceleration etc to see what effect it has on the servo.

MeJasonT, 2019-03-27:
[https://github.com/robottini/grbl-servo](url)

Ianmcmill, 2019-05-24:
@MeJasonT all your links redirect to https://github.com/grbl/grbl/issues/url or is that just me?

franchouille77, 2022-05-12:
I had the same problem with the `Sketch too big.` error on arduino IDE after enabling coreXY for my CNC plotter. For this I remove  a lot of code used for the spindle and coolant, and after what the sketch upload without problem. The code I made is here:
[https://github.com/franchouille77/grbl-light-version](url)