grbl/grbl #1520: Y axis not working after adding Z axis

bzflaglegomaniac, 2019-04-21:


Rob2029, 2019-04-21:
Have you tried to change the grbl version you are using? you did not mention your version and i think it is important, you can be using a modified grbl version for laser use or the 1.1 version that supports laser i think, anyway it also looks like a wiring problem or power supply problem since you said that the voltage drops on the power supply, that could be a power shortage. if the problem is only on the Y axis you could check its function isolated from the other axis or changing the driver from one axis to another

bzflaglegomaniac, 2019-04-21:
Thanks for the feedback. GRBL version is: [VER:1.1f.20170801:]

V1.1, as I understand it, changes the Z+ limit switch from an input to an output is used to toggle the laser. I still need that since half the point of adding the Z-axis was to raise and lower the focal point of the laser to facilitate cutting wood. It's a pity that semi-conductor lasers have such a high beam divergence as compared to gas lasers. With a gas laser this wouldn't even be an issue, but I digress....

The power supply is actually fine. Note that, as stated, if the X and Y-axis motors are run simultaneously (ie. a diagonal movement) both axis motors run fine even though running both motors obviously consumes more power than running one they Y-axis independently. 

As an experiment, since I'm revisiting the subject of using 2 motors simultaneously, I just tried running all three motors simultaneously (m1 x200 y200 z8) and it went without a hitch.

Again, though, this involves dividing the Arduino's time between sending commands to three different motors. When the Y-axis is used independently - and get's the Arduino's full, undivided attention - the ramp-up can lock up. It's intermittent. It's as though the timing is just border-line out of spec for my particular mix of equipment. 

Thank you for the suggestion about swapping the axes. It's the one test I hadn't thought of and it worked: that is to say, all axes now run smoothly, but it leaves me with a machine where X-axis is now the one moving away from me. I can live with that for now, but a proper fix - how to adjust the timing is really what's needed.

BTW. the voltage drop problem is not a symptom of a weak power supply (it can be in some cases, but not here). It's actually a symptom of two adjacent FETs being in the H-bridge that drives the motors, both being energized at the same time. This creates a dead-short to ground, albeit for a millisecond, which momentarily exceeds the power supplies source-current limits. Timing wise, the diagonally-opposite FETs in an H-bridge must be ON at the same time while the two in the other diagonal are OFF. You then switch off the two that were on, wait (because nothing happens in zero time), then switch on the opposite two. The problem with this is that it results in a motor that is constantly starting and stopping. It's noisy and it's inefficient. In a perfect world, the motors momentum would have carried it past the point where its rotor poles align with the fields of the internal permanent magnet at which point you energize the opposite pair of FETs to give the motors rotor the opposite polarity so that it now repels the permanent magnet and, now pushing, continues to rotate the motor in the same direction. If the timing is too slow, the motor loses momentum and essentially has to be continuously re-started which wastes energy, reduces average torque and is noisy. Tighten the timing up too much, and the motors poles can be reversed *before* alignment with the permanent magnet, causing the motor to try to reverse itself and appear to "lock up", which is what I was observing. Screw the timing up even further, and allow the second diagonal pair of FETs of the H-bridge to turn on before the previous pair has turned off, and you momentarily short out the power supply, which is also what I was observing.

It's a timing issue. I'm all but certain of it. Short of digging out my old oscilloscope (I'm a retired electronics technologist), the easiest way would be to simply tweak some software settings until the problem goes away.

bzflaglegomaniac, 2019-04-22:
Actually, I take it back. swapping the axes did *not* fix the problem. I just did a test laser cut and the Y-axis is still skipping steps. I'm going to try one as I write this without the Z-axis connected to see if that helps.....

The print was perfect without the Z-axis connected.

Rob2029, 2019-04-24:
too bad it wasn't that easy to fix, it sounds that the problem is a hard
one and i would consider to replace boards and drivers an experiment with
different hardware

bzflaglegomaniac, 2019-04-26:
Rob2029: your suggestion about replacing the stepper driver boards had me looking them up online with an eye toward also replacing the A4988 boards with DRV8825 boards. While I was looking at the pictures, it occurred to me that I hadn't tried adjusting the potentiometers to limit the drive current. I suppose that's the one other thing I could easily adjust before buying new hardware. Mind you, that was at 1 in the morning, so I'll try it out and report back in case anyone else has a similar issue.