grbl/grbl #1521: arduino UNO + grbl 1.1 + UGS

GastonMelo, 2019-04-24:
Hi to all, I download UGS from github https://github.com/winder/Universal-G-Code-Sender. Im using linux Tumbleweed 64bit, the board is arduino UNO + cnc shield + stepper motor NEMA17 + grbl 1.1 on arduino board.
The problem as I can see is the Y axe. If I push the +y button on the jog controller tab, the motor advance counter clockwise. If I push the -y button, the motor keep going counter clockwise (should go clockwise ). 
If I push two times the +x button, the motor first goes clockwise and then moved counter clockwise to full speed.  Here a pic of configs and version grbl. Thanks.

[image]

GastonMelo, 2019-04-25:
update: pin 6 arduino Uno board is keep at 0 volt

GastonMelo, 2019-04-25:
i found the problem, the pin AIN0 (that is connected to pin 6) of the atmega328 is missing.

GastonMelo, 2019-04-26:
can I modify the grbl to get the y direction by using other pin?

manojprasanga, 2019-10-05:
Yes.you have to edit grbl source code cpu map