grbl/grbl #1522: Steppers Motors Not Moving

Bugz2find, 2019-05-04:
I am completely new to this. Just want to throw that out there. My spindle is moving but when I load a program into the grbl controller 3.6.1 the steppers don't move. Previous to the beginning of the program they are humming and provide a resistance to me turning them manually. 

This is all i received from the manufacturer:

[image] 

[image]

https://cdn.fbsbx.com/v/t59.2708-21/57256984_550103998728975_4079442469832359936_n.doc/interface-description1.doc?_nc_cat=100&_nc_ht=cdn.fbsbx.com&oh=84e2971e3132987bc6fab1f6eb50f802&oe=5CCEF3E1&dl=1&fbclid=IwAR2I1ZGCRQvJkcVsHStbbwGAjg491LzantDoQIC7AtDVhb7Kg7Jz4RHmdhI

#	ASIN	Product Name
1	B079PNSVB1	Benbox DIY Mini 1310 Metal Engraving Machine Cutting Engrave PVC,PCB,Aluminum,Copper CNC Router
Message:
Hello friend
You can download grbl software from below link,and attached manual,thanks!

https://grbl-controller.software.informer.com/download/

install video can see from below link,thanks!

https://www.amazon.com/gp/product/B079PNSVB1

Kind Regards
Wendy

These are photos of the controller:

[image]

[image]

Bugz2find, 2019-05-04:
Also many of the other users of this engraver said they used grbl.. maybe I'm using the wrong program to control it with, but this is the only one I could connect to the controller with at all.

woshituzaizi, 2019-05-04:
I do not know if you can receive this mail
but first,if the program is not running,the spindle should should not moving too.
sencode,as usual，Red wire is positive，Black wire is negative，the picture`s Lower right corner is not good,
you should confirm to your manufacturer,
if it connected wrong,it might burnout the driver board。

https://cdn.fbsbx.com/v/t59.2708-21/57256984_550103998728975_4079442469832359936_n.doc/interface-description1.doc?_nc_cat=100&_nc_ht=cdn.fbsbx.com&oh=84e2971e3132987bc6fab1f6eb50f802&oe=5CCEF3E1&dl=1&fbclid=IwAR2I1ZGCRQvJkcVsHStbbwGAjg491LzantDoQIC7AtDVhb7Kg7Jz4RHmdhI

https://grbl-controller.software.informer.com/download/

https://www.amazon.com/gp/product/B079PNSVB1

Bugz2find, 2019-05-04:
Photos of controller:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

That's the only place for me to connect it. The only other place with the same connector is the DCout but that keeps the spindle continuously moving. 

Thanks for the response!!

Bugz2find, 2019-05-04:
I am not new to electronics I'm in school for electronics currently. So I will understand most of the talk. I just don't know about this particular field. I have also played with a raspberry pi and arduino uno.

Bugz2find, 2019-05-04:
@woshituzaizi

woshituzaizi, 2019-05-05:
Hi:
   i did not read all the source code,but i worry about S112 z = 0.000 (z max rate,mm/min)
please change it to 500.000 and try again
  check again,your wiring is good

Bugz2find, 2019-05-05:
I changed it to 150 earlier I’ll try the 500 also I have the Y axis moving a little bit at the beginning of a file. I was having trouble with undefined feed rate but I think when I changed the $112 it stopped. I think the driver current may just be too high.

Bugz2find, 2019-05-05:
Works great thanks I'm not sure if it was that or the potentiometer but one of those two worked great!

closed 2019-05-05 by Bugz2find