grbl/grbl #1526: limit switch repeatability

J-Dunn, 2019-05-13:
As part of testing my hardware I want to be able to do repeated homing cycles and print the difference applied to each axis each time. 

I set up and array storing sys.position[idx] at the top of limits_go_home() and added a printout to report.c but it always shows zero which I know is wrong. 

What variable do I need to be monitoring ?  
This would be a nice ( code cheap ) feature to add wrapped in #defines for hardware debugging. 

TIA.

109JB, 2019-05-13:
AFAIK, Grbl doesn't keep track of position (steps) during homing since homing by definition means it is finding its location by looking for the switches.

109JB, 2019-05-13:
I found @chamnit response to a similar question I had a while back and here is his response:

> @109JB : Grbl's homing routine is not compatible with this idea. During homing, Grbl doesn't track position, since it assumes it's undefined. It also doesn't bother with smooth decelerations upon detecting the limit. To do something like this would require a complete overhaul of the routine to behave more like the probe cycle routine (which is possible).

> At this point, I'm not interested in overhauling homing yet again and trying to cram it into v1.0 along with everything else. But, I'll add this to the development list for the next project. Even so, I'm not sure how useful this is, except in the case of setting up a machine. Once setup, it's likely this would only be used on occasion to help figure out problems with a job.

> That said, it's completely possible to do this same thing via probing or using a separate measurement tool, like a dial indicator. Wire in the homing switches, temporarily, to the probe pin, disable hard/soft limits, and probe the limit pin. The reported position will tell you the position relative to homed position. With a dial indicator, just fix it to somewhere in your machine, tell Grbl to move to it, and zero it out. During or after a job, move back to that same position. If the dial indicator is not zero, then the machine lost steps.

J-Dunn, 2019-05-13:
OK, that explains why I couldn't figure out how to do it!  Very helpful. 
I have a dial gauge , that will probably answer my question.  I was hoping to have an answer in motor steps since that is a digital answer from the driving mechanism, rather than analogue including mechanical play. I was hoping to isolate just switch repeatability. 

Messing with probe sounds like a valid solution but a bit of a PITA rewiring inside my boxed up electronics.

109JB, 2019-05-13:
Kind of a shaky video, but here is how I tested my limit/homing switches for accuracy.

https://www.youtube.com/watch?v=IGO47l62T2o&t=6s

J-Dunn, 2019-05-13:
Thanks, I've just put a dial gauge on mine and the good quality industrial switches are repeatable to within 0.01mm , that's about half a thou if you prefer dynes, slugs and ergs. 

Machine rigidity is about 5-10x that , so I'll stop worrying about the switches.  Well one was a dinky microswitch and that's going to get changed. This actually way better than I expected for switch repeatability. Really not an issue. ;)