grbl/grbl #1527: Arduino UNO GRBL 1.1f resets itself mid job.

Slixxor, 2019-05-16:
OK,
This is frustrating as hell.

I've setup many GRBL kits for our customers and havent seen issues quite like our demo unit in the office.

The issue:

Mid job, it stops, and doesn't receive an 'OK' back to the next position. At the same time, all values zero themselves.

Things we've done to combat the issue:

- Change Unos
- Applied a complex DC filter going to the board on both positive and negative sides of the DC input. We have physically tested with an oscilloscope and the DC input on both the DC ground and DC positive fluctuates within 0.01 millivolts.
- AC ground filter
- Changed Vout on DC input Vreg to 9.0v instead of 12.0v
- Opto-isolators fitted to limits to minimize any noise issues
- Ferrite sleeves fitted to USB cable, limit switch wire bundle and spindle wire itself to combat any EMI the VFD produces.

We are pretty much out of ideas.

Can anyone suggest anything else?

chamnit, 2019-05-16:
Please post your Grbl $$ settings and $I build info output. Also what are using for a GUI and is it the same particular job that fails or a random assortment of jobs.

Slixxor, 2019-05-16:
$I output:

Grbl 1.1g ['$' for help]
[VER:1.1g.20181112:]
[OPT:V,15,128]

Settings:

$0=10
$1=255
$2=0
$3=2
$4=0
$5=1
$6=0
$10=115
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=200.000
$101=200.000
$102=200.000
$110=2000.000
$111=2000.000
$112=500.000
$120=150.000
$121=150.000
$122=150.000
$130=300.000
$131=500.000
$132=200.000

Issue occurs at random times on random jobs, when running a failed job again it may succeed successfully 20% of the time and fail at a random time 80% of the time. 

Issue occurs in GRBL panel and Easel.

chamnit, 2019-05-16:
Ok. The settings and build look normal. Do jobs fail when you do a dry run? Meaning no spindle or even steppers powered?

Slixxor, 2019-05-16:
yes. have tried both dry and running.

I'm in the process of rewiring the whole thing in a larger enclosure with more space between the VFD and controller in case EMI is causing the issue.

There is no spindle control signals going to the VFD either. It's isolated from it.

chamnit, 2019-05-16:
It’s most likely an electrical issue. It almost always is. It’s completely possible it’s the the steppers, external noise sources, or grounding issues. 

Probably the first thing to do is a dry run on just an Arduino with the Grbl build and settings you are using on your machine. Nothing else connected. This will tell you if it’s something on Grbls+GUI side or machine hardware side.

whitetd, 2019-05-16:
also try a different USB cable.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Slixxor, 2019-05-16:
> also try a different USB cable. <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon> Virus-free. [...]

I will replace this in the panel aswell. i'm using a a USB-B to USb-B panel mount flylead. I've already tried swapping the main USB lead that plugs into it.

Slixxor, 2019-05-18:
OK, just an update. I have fully rewired everything and put it in a larger enclosure. No hardware been changed at all except the Uno (again). Same VFD, spindle, PSU, output wiring.

The only small change made was the neutral was connected to the 'S' on the VFD not the 'T'. It had a note on the bottom of the VFD saying "if operating in single phase mode to use R & T. Keeping in mind it still worked fine with 'S'.

Anyway, Ran a few jobs and it all seems good! Maybe that was the issue.

closed 2019-05-18 by Slixxor