grbl/grbl #1528: how to connect and run a VFD/spindle in GRBL 1.1

tkms002, 2019-05-16:
I have completed setting up my CNC router and have it all running nicely.  I am trying to find how to connect my UNO/Screwshield to the VFD so the G-Code/GRBL turns the spindle on/off and controls speed.
I have seen several posts that reference the WIKI but I have not been successful in finding any related info on the WIKI.
Can someone point me to a detailed explanation please?

Slixxor, 2019-05-18:
what VFD have you got?

tkms002, 2019-05-26:
it is a 3KW Huanyang

jahnj0584, 2019-05-26:
https://drive.google.com/file/d/0Bz73Ol-bOM8aOHo1aDM4Q3o5S0U/view?usp=drivesdk

tkms002, 2019-05-26:
excellent
Now  what parms do I have to change?
Thanks

jahnj0584, 2019-05-26:
https://youtu.be/7OszQS_BQMk

tkms002, 2019-05-27:
that is for linuxcnc not GRBL or am I not understanding something?

m0n5t3r, 2019-08-05:
* connect the ground to acm, pwm spindle pin to vin
* set the jumper for vin 
* set the vfd to be controlled by 0 to 5V
* devise some way to turn on the VFD (either hardwire it so it always runs clockwise, or do as I did, have to press run on the vfd before the spindle reacts to pwm - additional safety)

there is a [configuration option](https://github.com/grbl/grbl/blob/master/grbl/config.h#L271) to use the spindle direction pin as an enable pin if you want more convenience, too, but I was too lazy to recompile the thing when I got the VFD

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68