grbl/grbl #1529: Probe pin destroyed

mrguen, 2019-05-20:
Hello,

While I was probing the board the "PS" indicator got stuck on ":P". The impedance of the pin is now very low (360 ohms). I suspect the board was charged with electrostatic and it burned the A5 pin. 

So I was considering using another (unused) analog pin. I modified the "PROBE_BIT" number in cpu_map.h file . But now I see

PS:PR when using pin A0
PS:R when using pin A4

So apparently there might be more advanced probing functions implemented...

1) What are the signification of PS:PR and PS:R and probably other codes ?

2) Is there another analog pin that can be used ?

3) If it is not possible to use another pin I will need to change the Atmega328p, but how will I burn a bootloader on this since I don't see the ISP connector?

4) Finally I think a series resistor + an ESD protection diode might be good to add on these pins.
 
Sincerely

chamnit, 2019-05-20:
You can probably use A4. Don’t enable mist coolant (default disabled. If you don’t know what this is, ignore it.)

It’s pretty simple to flash Grbl on a new board. See wiki for details. Also go to the Gnea/Grbl site to get the most recent version. On front page.

mrguen, 2019-05-20:
I tried but apparently burned the chip.

mrguen, 2019-05-28:
Hello. Just to let you know that I 

-  replaced the Atmega328p

- added a series resistor on the z- probe line and a 6V zener diode to protect from another possible ESD discharge into A5 pin.

- used SPI to upload the booloader (Arduino Pro / 5V 16 Mhz) as explained in https://github.com/gnea/grbl/issues/336

- uploaded the sketch  as explained in https://github.com/grbl/grbl/wiki/Compiling-Grbl 

And it is working again. Apparently the A5 pin would need some more protection.

FelixWeichselgartner, 2022-03-02:
@mrguen Do you think I could use an optocoupler to protect the A5 pin?

mrguen, 2022-03-03:
@FelixWeichselgartner I suppose so. Usually it is used to protect from normal working condition high voltages like sensing on a 220 V circuit. I don't think it would be usefull in low voltages scenarii where you can simply use a series resistor plus a zener if needed.