grbl/grbl #1530: Spindle Speed Control

wheemzy1, 2019-05-20:
I set up the firmware and started playing around with the controller.

Motion seems fine but the spindle speed seems to be binary on/off rather than continuous.

I can't seem to be able to control the speed at all.

Either the spindle is off using the slider and then after moving the software slider more than halfway then it suddenly turns on full speed. I have a Dewalt router spindle and I'm using a SSR with a 10A fuse.

Any suggestions?

Thanks!

jahnj0584, 2019-05-20:
The router doesn't accept PWM so it's controlled via the dial.

A VFD used PWM or variable voltages to control the speed

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?
im using 2 stepper motors for x,y axis , solenoid for z axis , stepper motor as spindle
im really in rush for solving this problem
please