grbl/grbl #1532: Stepper motors not working Protoneer v3.51

alandroid, 2019-05-24:
Please someone help, i have lost many hours trying to get stepper motors to work and best i manged to get was holding torque but no movement. I seem to think i have incompatible hardware like wrong motors and my shield, arduino or stepper drivers are bad, so im not sure what to do, if i dont get help here i think i have to buy new electronics and hope i get lucky.

109JB, 2019-05-24:
$110 ,-$112 all equal 0. These are the maximum speed settings in mm/ minute. It can't move if 0 is it's fastest speed

alandroid, 2019-05-25:
I tried to change to:
`$110=300.000
$111=100.000
$112=100.000`
neither 100 or 300 make a difference but thank you very much for pointing that out, im sure it helps overal

chamnit, 2019-05-25:
Try resetting your EEPROM by issuing Grbl the ‘$RST=*’ command.

alandroid, 2019-05-25:
That had no effect, my motors still hold torque, make buzzing noise but dont move :(

chamnit, 2019-05-25:
Buzzing when you command it? That means you probably don’t have the current set high enough on your step drivers.

alandroid, 2019-05-26:
> Buzzing when you command it? That means you probably don’t have the current set high enough on your step drivers.

Only X axis works though, i swapped driver and motor.

terjeio, 2019-05-26:
Over to million steps per mm for Y and Z ($102 & $103) is perhaps a bit too much?

alandroid, 2019-05-26:
> Over to million steps per mm for Y and Z ($102 & $103) is perhaps a bit too much?

Wow, thanks for noticing that. Let me try again.

closed 2019-05-27 by alandroid