grbl/grbl #1534: Does GRBL use line numbers in any way?

fedekrum, 2019-07-02:
I am trying to implement a remote offline controller.
As in this device memory is critical for me, I am trying to parse the gcode and remove spaces, comments, empty lines.
Y was wondering if I can safely remove N1234 characters at the beginning of the lines for GRBL specific code.

Thanks

jahnj0584, 2019-07-02:
Yes. It's only for user reference. Most Gcode editors have a button to
add/remove line numbers.

All lathe codes i've seen dont use it because you never restart midway
through an operation and solely go by tool number.

fedekrum, 2019-07-02:
Thanks @jahnj0584  !!
Any other thing that can reduce code size that you can think off?

PicEngraver, 2019-07-04:
Remove repeating modal commands such as G1, etc.

Norwegian-Gadgetman, 2019-07-04:
Consider the accuracy of your machine, and how many digits of accuracy is in the commands.   
A G1 X15.12345 (in millimeters) is a waste of digits if your machine isn't capable of that accuracy.  
Also, many tools will generate positions for ALL axis even if it doesn't need them. 
If your tool is at X20, Y15,,Z10, a G1 X20, Y15, Z0 can be replaced with just the G1 Z0. 
(Assuming absolute positioning, of course. In Relative positioning any axis that reads o can be removed)