grbl/grbl #1535: Software baudrate selector (failing communications

Harms2019, 2019-07-16:
**Fixing serial communication**.

Hallo everybody,

Once upon a time, when I was a young student electronics, in the time when the baudrate of 110 baud was improved to the formidable 9600 baud, we used to build and repair RS232 serial devices ourself.  Without any in- and outside knowledge this was an almost impossible task.  The formost non catchable part of serial communication is noise and interference, caused by the neigbour's kitchenmachine, whether conditions as humidity, statics and lightning, family members using light switches and so on.  Without these factors the world would be perfect.  As times goes by, serial communications improved from RS232 to USB 3.2 at 16 Gbits/sec.  However, ... the non catchable part of serial communications didn't declare itself obsolete, but is still there to spoil the party.

Since grbl uses a default baudrate of 115.200 baud, more then once the communication suddenly breaks down, stops, kills itself, and many grbl-users are puzzled; "what did I do wrong?".

First of all; the first part is 'signal shielding'.  For the makers; prevent noise and interference from even reaching the signal lines by appropiate shielding.  This applies to all signal lines; switches, end swiches, every input signal (including the zero volt lines) must be shielded, and central 'star wired' connected to one single 'signal earth'.  An opened PC casing is not signal noise shielded anymore.  An open wired electronic web as a device on the table next to the CNC machine is often the same as building a 'noise antenna'.  

The second part is live itself.  The world is not perfect and shit happens.  Noise does often spoils the party.  Sometimes a bit tumbles over.  In serial communication the RX shift register is sometimes just too late to catch the signal raise.  The USB cable is too long, and interfering 'signal echoing' starts to appear.  The used USB port on the PC is too slow or to noisy (try another one).  The USB cable is not adequate shielded (try another one).  There are many reasons why serial communications breaks down.  grbl newbies are at first confused by the cable type; "I am using USB, not RS232!".  True, however, the Arduino's processor uses old school RS232.  Therefore a special USB-to-serial chip is added to the Arduino for using USB as an envelop to send RS232 signals.  Whenever the Arduino CNC machine responds (even with garbage) the USB-part of the communication is up and running.

Setting up the RS232 optimal baudrate is the third part.  In an industrial environment it is common practise to manualy raise the baudrate to the next setting, whenever testing the current communication speed went well.  At the moment a baudrate is set to high, communication breaks down, so the previous baudrate is the optimal one.  This speed-step-up-switch-down technique is by design embedded within the USB protocol itself, and garantees downward compatibility and highest possible speed settings; one can use an USB 1.0 device on an USB 3.0 port and vice versa.

By default grbl sets the minimal baudrate for RS232 communication to 115.200 baud for the reason of simulating Real Time machining (Keep in mind, simulating Real Time is not real Real Time at all!).  In a noisy or interference critical environment this default speed might be far to high.  Whenever serial communications  breaks down too often, it is recommended to lower the baudrate of grbl.  Unfortunately there is no easy simple way to achieve this.

An Arduino IDE software engineer downloads the grbl source code, opens config.h, locate the baudrate setting, alters this to a new baudrate value, compiles and uploads the revised grbl-software to the Arduino of the CNC machine.  

Not quite a simple task for the common 3D printer / lasercutter enthousiast.

**Feature request.**

In a previous era, serial equipment used to have a baudrate selector switch.  Changing the baudrate was then a simple turning a knob anybody could do.  Thinking it over; why not use a new grbl $-variable like   $B    to set the baudrate of the Arduino's FT232R chip?  Bij changing the $B value and a reboot everybody can simply change, tests and fix RS232 communication settings.

My hands on Arduino IDE experience is limited, and the $B-feauture should actualy been incorporated within grbl itself.  The following tested example gives an idea of how to prevent odd or wrong values for $B, and how a software baudrate-selector might be designed.

[grbl_baud-proposal.pdf](https://github.com/grbl/grbl/files/3396892/grbl_baud-proposal.pdf)

chamnit, 2019-07-16:
@Harms2019 : Yes it would be a nice to have. However, if you hadn’t noticed, the newest firmware completely packs the 328p’s flash and memory to its maximum. Small user-friendly improvements had to be omitted (still able to change through config.h and re-compiling). Don’t worry though, most of these will be addressed in the next major release (soon).

closed 2019-07-16 by chamnit

langwadt, 2019-07-16:
reducing the baudrate to try to fix interference is like putting a band aid on a broken leg

cri-s, 2019-07-17:
It is not really correct as the broken led could be the wrong baud rate.
Here are the values and error at 16Mhz osc.
The resonator adds annother +-1.1% error and the xtal adds +-0.15%
error on top of these numbers. 9bit transmission require +-1.5%
accuracy for industrial applications,
commercial applications allows +-3%, For a CNC, i tend to use the
industrial rules.
This calcs includes the aging over 10 Years and so on, like temperature error.
As you can see, 28.8k 57,6k and 115,2k are not allowed for reliable operation.
but other baudrates fulfill the allowable error and 28.8 is allowed if
xtal is used..

9600		14400		19200		28800
103	0.2%	68	0.6%	51	0.2%	34	-0.8%	

38400		57600		76800		115200	
25	0.2%	16	2.1%	12	0.2%	8	-3.5%

250K, 500K, 1M, and 2M are alternativly possible and supports depends
if the usb device
could handle thät baudrate nativly with what error. As example some
devices could handle
1Mbaud only with 2.1% error and as it consider comercial 3% error
range there indicate
that as supported even if the error is out of scope for industrial applications.

AlexHolden, 2019-07-18:
If you're using an Arduino Uno with the Atmega16u2 USB to serial device, both microcontrollers are generating the baud rate clock in the same way from a 16Mhz oscillator (presumably from the same batch), running at the same voltage and approximately the same temperature, so the relative error between them should be much less than the absolute error given in the datasheet.

cri-s, 2019-07-18:
Even then the evil is in the Details.
UNO R3 uses 8mhz xtal, the Main processor 16mhz resonator.
The atmega 16u2 have -3.5% baudrate error at 115.2k and grbl have +2.1%
error.
Total error is 5.6%. In case grbl configure the serial differently, the
overall error is 0 as if not using the U2X0 flag it archive -3.5% baud
error instead of +2.1% , always referenced to 115.2k serial speed.

Decreasing the speed in this case is one possibility of resolving this
problem.
Rs232 could work up to +-6% or 5.5% or 5% if you take into account other
error factors and have minimal margings.
However in this case every minimal noise/emi interference could result to
wrong reception as 8bits from 16x oversampling are wrong (5.5% and up).
And it works only because no parity and stop bit have 0 level.

And there exist other uno's that have real usb2rs232 chips.

Harms2019, 2019-07-18:
Well, as already stated the devil is in the detail.  Another little devil is that most baudrate calculations uses nice round values as 300,  600,  1200,  2400,  4800 and 9600 baud.   However, if an 8 or even 16 Mhz xtal is used, the higher the required baudrate the more difficult it becomes to generate the correct baudrate.  Calculating the catching bandwidth, one does have to take in account the real baudrate values, such are for the FT232R, FT2232 and FT232 in theoretical reality 19230   , 38461  , 57692 ,  115384  , 230769 ,  461538  and  923076  baud.  Keep in mind; this mismatch is on the Arduino side.  Unclear is the generated baudrate mismatch on the PC-side where the RS232 signals are enveloped within USB packets.  Thinking it all over (especialy for 2 meter or longer), it might even be a much better idea to skip the USB cable wiring all together, and turn back to a good old RS232 cable via a MAX232 connected to the Arduino.

AlexHolden, 2019-07-18:
> Even then the evil is in the Details. UNO R3 uses 8mhz xtal, the Main processor 16mhz resonator.

Are you sure? I thought both microcontrollers on the Uno were clocked at 16Mhz. That seems to be what the official Arduino schematic shows. Since they are both AVRs with the same type of serial peripheral, even if there is a few percent theoretical error in the baud rate generator, both devices should have exactly the same error.

AlexHolden, 2019-07-18:
> Well, as already stated the devil is in the detail. Another little devil is that most baudrate calculations uses nice round values as 300, 600, 1200, 2400, 4800 and 9600 baud. However, if an 8 or [...]

The serial transfer (which technically isn't RS232 because it happens at 5V logic levels) only takes place between two chips mounted on a PCB about an inch apart. The USB side of the link has nothing to do with the baud rate setting. Do you have any evidence that the data is getting corrupted over the connection between the two chips on the Arduino? If so, are you arc welding next to your CNC by any chance?

cri-s, 2019-07-19:
Yes, both version exists, newer version with smd 328 have 8Mhy Xtal.
Internally, both works with 8Mhz clk (even if F_CPU is set at 16MHz) .
The versions with 16Mhz have the div2 clock divisor enabled.
On 115.2K it uses U2X0 , so effectivly it is the same as 16Mhz but
with U2X0 cleared.
Grbl at that speed always uses U2X0 set. This is 5.6% difference.
The Stop bit starts from ust to uC at 75390 ns  and the uC recognize
the same edge at
79766 ns.  One bit is 8681 ns. This is more then half of the error.
For the last bit transmitted, it starting at 67013 ns and ending at 75390 ns.
Instead the receiver read it    starting at 70903 ns and ending at 79766 ns.
This means that on 8bit at supposed 16x oversampling, the valid
bitpattern is 8 from the 16 samples. One emi/noise spike and the data
is corrupted.  Flipping form 0 to 1 is protected, but flipping from
bit 1 to 0 is highly possible. As LSB is send out first, and
MSB is normally always zero,  it requires a spike or multiple spikes
that are 2.2uS long or
problems with the ground loops. This means it should work, but is not
stable as when
lower speed or annother arduino board or no 2x doubler is used.
But don't expect the solution works for 10-15 Years, after 7 Year the
aging of resonator
have changed the timing that it surely don't work anymore.

Harms2019, 2019-07-19:
OK, Sorry guys for throwing a bat in the chicken house.  

The communication between the Arduino's processor and the USBtoRS232 seems for me not to be the bottleneck.  However, if the Arduino is set at 115200 baud, the real value with an FT232R chip is 115384 baud.  Many people however do not realise USB is actualy a 'serial RS232 Next Generation', and has it own design flaws.

A modern PC is much much much faster as the Arduino's 8 or 16 Mhz base frequency.  The grbl GUI on the PC uses a RS232 terminal application being supposed to been set at the same speed as the Arduino.  If on the PC-side the RS232 signals are software build based on exactly 115200 baud, there is a mismatch of 184 baud.  This eventualy mismatched RS232 signal (-a letter-) is handed over to an RS232toUSB-application that 'envelopes' the software signals within several subsequent USB packets.  This OS dependable USB application ('device') is OS independable soft- and hardware connected to the USB Host chip (EHCI, UHCI, ...) where it is treated as one of the 127 possible USB network clients.  At the time the USB Host finds it fitted, it will send the enveloped signal packets to the USB client 'Arduino' (or 'CH340' or 'ACM0' or whatever name is used), where the USBtoRS232 chip will extract the software packets and rebuild its content as voltaged hardware signals.  The FT's build in UART's Rx- and Tx-shiftregister can process a certain amount of mismatch; called the catching bandwidth  (See the FT232 extended datasheet for more information.).  Suppose the PC's Tx and Arduino's Rx are functional within exact the same percentual small band width; as long as the Rx catches on the low side, where as the 'Tx sends' at the high side, communication is up and running.  If one or both modal Tx and/or Rx base frequencies shifts, often caused by the side effects (spikes) of interference or noise, communication is lost.  In a perfect world calculations (as been discussed) will give an answer.  In the real world however the not calculatable unknow interference factor often spoils the party.

Thats one part.  Another issue is USB's hardwired T-network itself.  To keep track on all the connected USB client devices, the USB Host will regularly pauses the on going packets sending, to check what USB devices are on line; this features makes USB 'hot plugable' (but also interrupts running communications).  With every increase of the 'receiving baudrate' the shorter this 'checking device interrupt' is allowed to be.  The more USB devices are connected to the same USB Host, the lower the effective packet speed will be.  Since 127 devices are allowed; why then does my PC owns 7 different USB Hosts chips?  Why would that be?   "Out of sync errors" experienced by many Arduino IDE users, often can be solved to disconnect all USB devices (use a wired keyboard and mouse), and test uploading to the Arduino at every available USB port.  In calculated lab conditions all those USB hosts circuitry conditions should be the same; in the real world thats turns out to be not true at all.

On the USB side the Arduino identifies itself as an 'USB 1.01 Class 2', meaning the host-client technique is a Master-Slave protocol.  There is no way the FT232 might send a 'data request' by itself.  Having no access to an communication interrupt at any way, the Arduino might easely 'times out'.  Therefore, save the best USB Host as the single host for the Arduino.

Last but not least; a very unpleasent unsuspected gnome is 'energy saving'.  Some 3D printer jobs might take more then 13 hours to complete.  With such a machining time duration a baudrate of 110 baud will do.  The common overall baudrate of 9600 baud is fast enough for almost any CNC machine.  Why then speed up the default baudrate from 9600 to 115200?  To picture this, it is like a peddling on a bike from 9,6 km/h and speeding up to 115 km/h.  To quote mr Armstrong "... not a small step, but a giant leap ...".  Avoiding unwanted disruptions, one tends to leave the PC untouched after starting a CNC job.  If the PC's energy saving is not switched off, well then after a while is will switch into energy savings mode, and this might be result in speeding down the PC and USB, or even to a suspended hibranation (deep sleep).

Going back to where this started.  ...  To avoid being intoxinated or dying by inhaled dioxines, I had to place my laser cutter at a more (-for me-) healthier location.  I had to use three 2 meter USB cables.  Yes, this is begging for trouble, and soon to often communication broke down.  The cause was quickly spotted and a solution implemented.  Fixing my problems, I noted the extremely raised baudrate in comparison with former grbl versions, and the enormous effort one has to undertake to simply change a baudrate-value.  Peeking, pooking and nosing around on the internet, I found to many grbl and Arduino users with communication issues and there efforts to fix it.  "We have tested this setting and found no problems".  Sure, a formula one race car sets massive speed records on a formula one race track.  In the real world however people tends not live on race tracks, but on small bumpy putholed roads.  If the paperboy tries to improve the delivery speed by using a formula one race car, well, thats up to him.  But don't come to me for complaining that car is stucked, broke down or hitted a wall.    "I just only want my daily paper; no more no less".

Thinking it all over, I made the suggestion to the grbl team to add a new   $B  parameter as baudrate-settings and chancing init's Serial.Begin([baudrate])  into Serial.Begin($B) .  Every grbl-user can simply change the grbl's baudrate very easy if necessary.  To prevent losing contact by entering invalid values, I also added a small checking routine.  No big deal.

Thats all folks!

P.S.   If I break my leg and bones are sticking out of the skin, I prefer a bandaid as soon as possible to prevent future amputation.

cri-s, 2019-07-19:
The arduino on GRBL source code send/read at 117647 BAUD, the FT232R chip at
115385 baud (3000000/26)  The error is 2%, this is ok.

For longer cable, use usb to ttl converter and rs422 , example SN75176
for driver and receiver and network cable or twisted cable. In case
with lm324 you could construct such
a driver if price or deliver time is a problem. On network cable you
can deliver the power too
if you need that.

I had done something similar, default is 19200, on 16Mhz, 9600 on 8Mhz
1200 on 1Mhz and 24400 on 20Mhz ., but i have stored it on separate
config system memory. inside eeprom that defines some configurations
using 3 bits.
.
I have implemented 19200 38400 57600  for 20Mhz
I have implemented 19200 38400 57600 2000000  for 16Mhz
I have implemented 9600 19200 28800 for 8Mhz (internal osc)
I have implemented 1200 2400 19200  for 1Mhz  (internal osc)

#define F_CPU 16000000
static void
uart_0(void)
{
#define BAUD 19200
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_1(void)
{
#undef BAUD  // avoid compiler warning
#define BAUD 38400
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_2(void)
{
#undef BAUD  // avoid compiler warning
#define BAUD 57600
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_3(void)
{
#undef BAUD  // avoid compiler warning
#define BAUD 2000000
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
#define F_CPU 20000000
static void
uart_4(void)
{
#define BAUD 19200
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_5(void)
{
#undef BAUD  // avoid compiler warning
#define BAUD 38400
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_6(void)
{
#undef BAUD  // avoid compiler warning
#define BAUD 57600
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}
static void
uart_7(void)
{
#define F_CPU 1000000
#undef BAUD  // avoid compiler warning
#define BAUD 19200
#include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
#if USE_2X
UCSRA |= (1 << U2X);
#else
UCSRA &= ~(1 << U2X);
#endif
}

void serial_ini(char i)
{
  if(!i--) uart_0();
  if(!i--) uart_1();
  if(!i--) uart_2();
  if(!i--) uart_3();
  if(!i--) uart_4();
  if(!i--) uart_5();
  if(!i--) uart_6();
  if(!i--) uart_7();

  // enable rx and tx
  UCSR0B |= 1<<RXEN0;
  UCSR0B |= 1<<TXEN0;
	
  // enable interrupt on complete reception of a byte
  UCSR0B |= 1<<RXCIE0;
	
  // defaults to 8-bit, no parity, 1 stop bit
}

void serial_init() {
	serial_ini(eeprom_ver()&7);
}