grbl/grbl #1536: My grbl uploaded failed  I already tried it more than 30times

AwJinLiang, 2019-07-22:
In file included from C:\Users\PC\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\PC\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\PC\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:123:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

Sketch uses 29868 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. Maximum is 2048 bytes.

Arduino china board uno with Ch340 driver, window 10
Most of the video in YouTube is helpless, can't solve my problem.
I have 0 knowledge with arduino
Thanks in advance if anyone can help me out of the problem.

chamnit, 2019-07-22:
This usually happens when you aren’t using the GrblUpload.ino example as downloaded. Like the example comments state, don’t alter it.

AwJinLiang, 2019-07-22:
I am using GrblUpload and never alter it.

chamnit, 2019-07-22:
What version of Arduino IDE are you using? And do you have any other copies of Grbl in your Arduino library path?

AwJinLiang, 2019-07-22:
Arduino 1.8.9
My grbl is in  C>User>Pc>Documents>Arduino>libraries (grbl is adding through add zip library in arguing Ide)
C>Program Files(x86)>Arduino>libraries (This is the location of where my arduino and others example such as blink... and default libraries are)

I have two grbl file another is under grbl_master which is in This pc >downloads

chamnit, 2019-07-24:
@AwJinLiang : Checked Arduino 1.8.9. I do get these compiler warnings, but it completes and uploads fine. Open your serial port and change to 115200 baud.

chamnit, 2019-07-24:
Also see this statement on the compiling [wiki page] (https://github.com/gnea/grbl/wiki/Compiling-Grbl): 

```NOTE: If your environment is clean and Arduino IDE compiler still throws "warning: [...] redefined" messages, you may need to uncheck File -> Preferences -> "Aggressively cache compiled core".```

closed 2019-07-24 by chamnit

AwJinLiang, 2019-08-09:
I used another laptop it still having the same error. I followed all the steps from https://github.com/gnea/grbl/wiki/Compiling-Grbl. Is this because of grbl file problem or arduino software problem or UNO board problem? 
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\danso\Downloads\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\danso\Downloads\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\danso\Downloads\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\danso\OneDrive\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:123:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

Sketch uses 29868 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1484 bytes (72%) of dynamic memory, leaving 564 bytes for local variables. Maximum is 2048 bytes.
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AwJinLiang, 2019-08-09:
After unchecked  "Aggressively cache compiled core", I get the error below
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

FindFirstFile C:\Users\danso\AppData\Local\Temp\arduino_cache_900580: The system cannot find the file specified.

Error compiling for board Arduino/Genuino Uno.

This report would have more information with 
"Show verbose output during compilation"
option enabled in File -> Preferences.

Start from AppData folder >>>>>AppData\Local\Temp\arduino_cache_900580 I cant find AppData folder under my Users\Danso