grbl/grbl #1538: Z axis not working while X and Y makes loud noises

NghiDao2, 2019-07-27:
So im making a 4xidraw machine. The x and y axis works normally, however they make really loud noises. However, the z axis doesn't work and just shakes while the X and Y axis motors made some noises. I tried unplugging either the x or y axis stepper motor, and the z axis starts to move a bit, if a unplug both the X and Y stepper motor, the z axis starts to move normally.
I used a 12V power supply, so i dont think thats the problem.
Can someone help me to get the Z axis moving with the X and Y axis plugged in ?

Thanks

closed 2019-07-27 by NghiDao2

jahnj0584, 2019-07-27:
You don't have enough current available. 12V isn't a lot either. I consider
24v the bare minimum for nema23's but prefer 48-60.

NghiDao2, 2019-07-27:
ok thanks