grbl/grbl #1540: Uploading grbl to uno from RPI 3 B+

johanndavel, 2019-08-06:
Hi guys
Im running Raspbian Buster with desktop on a RPI 3 B+, ive installed the latest arduino ide with the terminal and simple sketches upload to my uno.
When i try to load https://github.com/grbl/grbl as a library it just throws errors. now before i write an essay of error code and everything i have tried. mabe someone has managed to load grbl to uno from raspian. any help would be apriciated

chamnit, 2019-08-06:
Haven’t tried it from RPi directly but errors are typically due to multiple Grbl libraries in an Arduino path, not using the GrblUpload example, or having the agressively cache option enabled in the IDE preferences.

johanndavel, 2019-08-06:
My bad, you dont install it with the grbl folder, you use the GrblUpload example .ino file

johanndavel, 2019-08-06:
but thanks. il be lurking here for a while asking dumb questions haha

closed 2019-08-06 by johanndavel