grbl/grbl #1543: Laser CO2 PWM connection

CharlesMatt, 2019-08-22:
Hi !

Can anyone help me?

I have a 40w laser tube power supply with this connections and characteristics
[image]
[image]
and I did the connections with one arduino mega 2560 without ramps or cnc shield and the software that I use is LaserWeb, the nema motors are working, my problem is the PWM, this thing are not working and when I pulse the botton "Laser test" nothing happens, I based my connections in this image

[image]

and I repeat, all axis are working, the GRBL version that I use in this proyect is the 1.1, the only problem is the laser control or the pwm.

Help me please ! :C

AleDri, 2020-01-07:
I have the same problem. The laser's electrical source is always on, the PWM control does not turn off the source.

Help Please! Thank you.

CharlesMatt, 2020-01-23:
I fixed the issue, you have change the parameter 32 in the grbl config, and in the software create the next config with the parameters of the images
[image]
[image]
[image]

Save and upload the settings, more details here:

https://www.youtube.com/watch?v=oXHmOdObTvw&t=75s

hkuwabara3, 2021-08-25:
Laser power can be set to $32=1 $30=1000

LaserGRBL config
M4 laser on
M5 laser off
S: 0 to 1000
[image]