grbl/grbl #1544: Resetting with Ctrl-X

rafaluta1, 2019-08-22:
I am testing Grbl CNC Shield with my Arduino Uno. Everything is working except for the ctrl-x command. I tried on the Arduino IDE serial monitor but it appears that it dont recognize. I Tried:
Holding Ctrl with C and nothing happend.
I Tried sending ^X or ^x but the console responded "Error: Excepted command Letter" 
I Tried sendind 0x18 on the consoled and it responded "Error: Bad Number Format"

Does anyone how to solve this?
Any help would be appreciated.

109JB, 2019-08-22:
You can't do it in a text format. You need to send the special code that is associated with ctrl-X.  In a visual basic program you would do something like:

SerialPort1.Write(Chr(24))

So it can be done programatically, but not in a text editor or a typical serial port comminications monitor.

rafaluta1, 2019-08-22:
Ah ok. So i'll put it in my C# Code that I'm doing here. I'll Try it now. Thanks

rafaluta1, 2019-08-22:
Ok i Have here a C# Program wich I've made a "console" to type the commands, and I tried like this:
 if (txtEnviar.Text == "^X" || txtEnviar.Text == "^x")
            {
                portaSerial.getPort().Write(char(24));
            }

But The .Write in C# use a String as parameter, so I have to convert It to String?

109JB, 2019-08-23:
ctrl-X is a special ascii character.  "^X" doesn't do it, it is not able to be represented by readable characters.  Char(24) in VB is the ascii code for it.

terjeio, 2019-08-24:
From [my c# code](https://github.com/terjeio/Grbl_CNC_Controls/blob/master/CNC%20Controls/CNC%20Controls/Comms.cs):

```
        WriteByte(24);

        public void WriteByte(byte data)
        {
            this.SerialPort.Write(new byte[1] { data }, 0, 1);
        }
```

rafaluta1, 2019-08-24:
Thanks man. That worked!! Found the respective code in C#. For someone who needs it:
string s = Encoding.ASCII.GetString(new byte[]{ 24 });
SerialPort1.Write(s)

closed 2019-08-24 by rafaluta1

rajpal84-bit, 2020-05-20:
cnc_ard = serial.Serial('COM11',115200)  
y= "\n"
cnc_ard.write(y.encode())

This will help with python code. (To write ctrl-x)

Hubert880411, 2020-09-14:
> You can't do it in a text format. You need to send the special code that is associated with ctrl-X. In a visual basic program you would do something like:
> 
> SerialPort1.Write(Chr(24))
> 
> So it can be done programatically, but not in a text editor or a typical serial port comminications monitor.

Hi,John.I use your method:First, in config.h you need to change the following line for the new character

#define CMD_SAFETY_DOOR 0x92 ---> #define CMD_SAFETY_DOOR '&'
then in serial.c, case CMD_FEED_OVR_COARSE_MINUS: system_set_exec_motion_override_flag(EXEC_FEED_OVR_COARSE_MINUS); break;
just move this up about 3 lines to where you see the line starting with
case CMD_FEED_HOLD:
And I type '&' on Arduino serial monitor when g code is running.But it appears error1 on monitor.Please tell me how to trigger feed override command.Thanks John

Hubert880411, 2020-09-17:
> cnc_ard = serial.Serial('COM11',115200)
> y= "�\n"
> cnc_ard.write(y.encode())
> 
> This will help with python code. (To write ctrl-x)

Excuse me,if I use python send g code to grbl like this.
import serial
import time

# Open grbl serial port ==> CHANGE THIS BELOW TO MATCH YOUR USB LOCATION
s = serial.Serial('/dev/tty.usbmodem1811',115200) # GRBL operates at 115200 baud. Leave that part alone.

# Open g-code file
f = open('grbl.gcode','r');

# Wake up grbl
s.write("\r\n\r\n")
time.sleep(2)   # Wait for grbl to initialize
s.flushInput()  # Flush startup text in serial input

# Stream g-code to grbl
for line in f:
l = line.strip() # Strip all EOL characters for consistency
  print 'Sending: ' + l,
  s.write(l + '\n') # Send g-code block to grbl
  grbl_out = s.readline() # Wait for grbl response with carriage return
  print ' : ' + grbl_out.strip()

# Wait here until grbl is finished to close serial port and file.
raw_input("  Press <Enter> to exit and disable grbl.")

# Close file and serial port
f.close()
s.close()
Is it possible also send real time command(0x18) when g code is running?

luisDias89, 2022-10-11:
In Python the comand 0x18 is b'\x18' !!

luisDias89, 2022-10-11:
b'\x18'

QGB, 2023-06-08:
how to send cancel jog  ctrl-c?

hamidmohebbati, 2023-12-28:
hi.
i was wondering to know is it possible to add break in axes? to block moving in idle situation. especially in z.
thanks for your support.