grbl/grbl #1546: Problem with m3 on grbl 1.1f

delphy38, 2019-08-22:
I just received a CNC 3018 with a motherboard controlled by an arduino nano and grbl 0.9j, M3 works pretty well: the spindle motor turns/stops, for 30 seconds then starts finally.
I flashed Grbl 1.1f and now M3 doesn’t work at all.
With Grbl 0.9j I use grblControl_0.8.1.
With Grbl 1.1f I use Grblpanel-Release-1.0.9.18A.
I am new in the field of Grbl, is there a setting or configuration to run M3 on Grbl 1.1f ?
Thank you.

jahnj0584, 2019-08-22:
I believe the pins changed

chamnit, 2019-08-23:
Comment out `#define VARIABLE_SPINDLE` in config.h. Make sure you are editing the files in the library folder.

delphy38, 2019-08-23:
Merci.
Je l'avais déjà fait, avant d'installer la bibliothèque, et de téléverser grbl sur l'arduino.
Donc il y a un autre problème, apparemment.

delphy38, 2019-08-23:
> Je crois que les épingles ont changé

delphy38, 2019-08-23:
Why was this number closed?
I still hope to have some answers.

chamnit, 2019-08-23:
You closed it.

delphy38, 2019-08-23:
Non !

chamnit, 2019-08-23:
Yes.

chamnit, 2019-08-23:
@delphy38 : If you would like help, you need to provide more detail than saying it doesn't work like it used to. One way to check if you disabled VARIABLE_SPINDLE correctly is providing the $I build info output. It should not contain the letter V.

delphy38, 2019-08-23:
****Part of my config.h*****
// Sets which axis the tool length offset is applied. Assumes the spindle is always parallel with
// the selected axis with the tool oriented toward the negative direction. In other words, a positive
// tool length offset value is subtracted from the current location.
#define TOOL_LENGTH_OFFSET_AXIS Z_AXIS // Default z-axis. Valid values are X_AXIS, Y_AXIS, or Z_AXIS.
// Enables variable spindle output voltage for different RPM values. On the Arduino Uno, the spindle
// enable pin will output 5V for maximum RPM with 256 intermediate levels and 0V when disabled.
// NOTE: IMPORTANT for Arduino Unos! When enabled, the Z-limit pin D11 and spindle enable pin D12 switch!
// The hardware PWM output on pin D11 is required for variable spindle output voltages.
// #define VARIABLE_SPINDLE // Default enabled. Comment to disable. ************************

// Used by variable spindle output only. This forces the PWM output to a minimum duty cycle when enabled.
// The PWM pin will still read 0V when the spindle is disabled. Most users will not need this option, but
// it may be useful in certain scenarios. This minimum PWM settings coincides with the spindle rpm minimum

****** I will try tomorrow the order $ I build info

delphy38, 2019-08-24:
Good morning,
Here are the results of $I build info:
[image]

[VER : 1.1f . 20170801 : ]
[OPT : , 15 , 128]

ygjfggfgfj, 2020-08-28:
hello there
i dont have any programming knowledge
could any one help me how to set stepper motor as spindle in GRBL library?
im using 2 stepper motors for x,y axis , solenoid for z axis , stepper motor as spindle
im really in rush for solving this problem
please

delphy38, 2020-08-29:
Bonjour,
Je n'ai également aucune connaissance en programmation, et je n'ai jamais reçu de réponse à ce problème, qui me permette  de le résoudre.

closed 2020-08-29 by delphy38