grbl/grbl #1550: Read values from PRB after probing?

BernardG, 2019-09-06:
When I probe with my macro, I see that GRBL answers with PRB values:
[image]

Is there a possibility to read theses (from what I can see, they are absolute positions at the probing point detected) in G-Code?

MeJasonT, 2019-09-06:
send a ? character
and the reply message will give you both the work and machine coordinates

<Alarm|MPos:0.000,0.000,0.000|Bf:15,127|FS:0,0|WCO:0.000,0.000,0.000>
followed by ok

They are the machine coordinates in negative space,  Grbl has adopted the negative coordinate system used by industrial machines and has been written in such a way as to make proper machinists out of us and not allow us to make a fudge.

Im not quite sure how grbl handles the G38 command with regards the measurement storage and display of the position data, obviously its use is to zero the work axis.
which then raises a question id like an answer to is the probe output saved against the selected work coordinate ie G54, G55 etc  As far as i understand things the .2 on the end of the  command G38.2 etc is the work coordinate we wish the setting to be saved in.

BernardG, 2019-09-06:
Well thanks for your answer, but that was not my question. I asked if I could read those values in GCODE.

BTW, yes, the probe output is saved in the current work space (G54, etc..), and NO the .2 in G38.2 has nothing to do with the work coordinate, but with how the probe event is triggered. With G38.2, the  probe is triggered by the contact closed, while with G38.4, the probe is triggered by the contact opened.

If you look at what I posted, you can see at first G38.2, with a motion going toward the Y axis, and a value in PRB for Y of -222.232. Then you can see a 2nd G38, but G38.4, with a motion going AWAY from Y, slowly. As the contact was closed the first time, I check when it open again. And you can see it's worth it, as there is a difference of  97/1000 of a millimeter. PRB gives -222.195.

BTW, GRBL does nothing with G38. It's G92 Axis Value which will actually save something, Like G92 X-10, for example.

chamnit, 2019-09-06:
Grbl reports the step position where it first detects the appropriate pin change. The difference you see is due to something in your setup. 

Grbl just reports the probe results. It’s up to the user to do something with it. I think what you are asking for is if Grbl supports macro programming. The answer is no but certain GUIs like bCNC can emulate it.

MeJasonT, 2019-09-06:
I sat and looked up the command in between grouting tiles in my bathroom.  But the correction to my response was much appreciated.  I was also replying on the other thread that Sonny closed down in front of me.  We get a lot of stick for GIT being crap.  If the emails comes to our inbox and we respond to it, we actually have no idea there is another thread on the ether, the duplication advice is sound but always catches me out by having best intentions.
In response to your reply Bernard. Simply no there isn't a grbl gcode command to read these coordinates, unless Sonny has hidden one somewhere which does not meet with the standard used.  The grbl top level commands will provide the information you want try $# then use G53 moves to those coordinates (negative work space/machine coordinates)

chamnit, 2019-09-06:
@MeJasonT : Sorry about that! Just trying to keep answers in one place.

MeJasonT, 2019-09-06:
No Probs, i'm just a dinosaur living on the old GRLB GRBL site

MeJasonT, 2019-09-06:
now that's commitment
[image]