grbl/grbl #1552: How send verbose string output

manojprasanga, 2019-09-29:
Hi all,
         I'm looking for grbl verbose out put send from one(grbl running Atmega328) to another Atmega328 microcontroller via serial connection.
       How can do this?
       How edit grbl source code?

closed 2019-10-01 by manojprasanga

MeJasonT, 2019-10-01:
are you just looking to receive the output from one as an input to the second.
for both to transmit to a PC as well is near impossible.  To use the second to receive only is called a sniffer.  Just jumper the firsts TX line to the RX of the second (don't forget the gnd).  You have mentioned verbose which is the output of GRBL, you will not get to see the transmitted signal from the PC.  There are tools out there called virtual serial ports emulators VSPE which will allow you to do nifty tricks with serial ports, this is software running on a PC which can intercept both RX and TX communication simultaneously.

manojprasanga, 2019-10-01:
Thank you sir.It's work