grbl/grbl #1555: Mega 2560 + ramps 1.4 + tb6600 + grbl = false. Pls help!

leonhuynh2006, 2019-10-03:
Hi all pro,

I am new in CNC. I have connected mega + ramps + tb6600 as enclosed photo.
[image]

I used grbl 1.1 mega 2560 version, selected the right port (COM 3), the right board (mega 2550) in Arduino IDE. Uploading firmware was ok. 

But when I tried to test it by using Universal GCode Sender manually, but the motor didn't spin

Did I miss out something. Pls helpppppp

Thanks

MeJasonT, 2019-10-29:
please ask your question here
https://github.com/gnea/grbl/issues
this is the old site

It looks like you have not sent the command to unlock grbl $X

take a look at https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

normal practice would be to home your machine first $H which on completion disables the alarm lock.
If you had an expensive haas machine connected you probably wouldn't want it starting up and thrashing around with no idea where it was so by default grbl has the same built in protection by starting up locked.

if your motors are buzzing but not moving after that then the wiring on the motors could be the wrong way around, more info on the other forum - question has been asked a few times.
best of luck

mikmikmedia, 2020-03-23:
Hi, why would you use RAMPS if you are conecting external driver like TB6600? Conect directly on Mega like on wiki for grbl Mega https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega.

thawkins, 2021-03-27:
> Hi, why would you use RAMPS if you are conecting external driver like TB6600? Conect directly on Mega like on wiki for grbl Mega https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega.

The ramps card makes plugging in displays, endstops, servos (bltouch) etc much easier. There is a huge ecosystem of parts designed to plug straight into a ramps card.